CloverHackyColor/CloverBootloader #5: Boot iso image possible?

benneq, 2019-10-13:
As far as I've seen, Clover has an iso9660 driver. Is it possible to load / boot an iso CD/DVD image? I couldn't find anything on google / wiki.

roddy20, 2019-10-13:
some time ago, I made а Hybrid CD with Clover + macOS Recovery Image
and it boot successfully 
it may vary depending of OS/Clover version, I did not check it every day
but its is possible

benneq, 2019-10-13:
If you could provide some more information, I maybe could test it  :)
I already have no clue where do place and how to reference the iso files.
I can't put them on the EFI partition (i guess?), because this partition is much too small for most bootable images.

roddy20, 2019-10-13:
look at makeiso script included into Clover sources

roddy20, 2019-10-13:
the main command is 
`hdiutil makehybrid -iso -joliet -hfs -hfs-volume-name ${CDLABEL} -eltorito-boot ${CDBOOT} -no-emul-boot -ov -o "${ISOIMAGE}" ${IMGROOT}`

and place cdboot and image content to ${CDBOOT} and ${IMGROOT}

benneq, 2019-10-13:
Ah, okay. I think you misunterstood. I want to make a USB stick based on Clover, that can boot multiple ISO files.

So... you boot from the USB stick. Then you see the usual Clover GUI, where you can select your operating system. And it should show additional entries for my ISO images.

roddy20, 2019-10-13:
Yes, misunderstood. I made real bootable DVD disc, not USB stick with image

SergeySlice, 2019-10-14:
macOS installer do this.

closed 2019-12-30 by SergeySlice