CloverHackyColor/CloverBootloader #6: Clover Package doesn't work on Catalina

bpresles, 2019-10-14:
On Catalina (OS X 10.15), Clover installer can't get read/write access
[image]
[image]

Of course I've SIP completely enabled (0x0) ;). I didn't try without SIP, but I don't want to have to disable it each time I need to update Clover

esetnik, 2019-10-15:
I have the same issue

riverLethe9, 2019-10-18:
I have the same issue. It does not seem to be version specific.

simmy1232, 2019-10-18:
Can install but the system crashes. It's also made my system very slow

vectorsigma72, 2019-10-19:
> On Catalina (OS X 10.15), Clover installer can't get read/write access
> [image]
> [image]
> 
> Of course I've SIP completely enabled (0x0) ;). I didn't try without SIP, but I don't want to have to disable it each time I need to update Clover

Hi, what happens if you write your administrative account name and your administrative account password?

bpresles, 2019-10-19:
It shows the error shown in the screenshot:

https://user-images.githubusercontent.com/220623/66786465-b0ee4100-eee0-11e9-96c2-312c2831f27d.png

vectorsigma72, 2019-10-19:
> Same here.
> 
> ```
> g3nsvrv@G3NSVRVs-Mac ~ % sudo mount -u -o rw /Volumes/Install\ macOS\ Catalina 
> g3nsvrv@G3NSVRVs-Mac ~ % sudo installer -pkg ./Downloads/Clover_v2.5k_r5096.pkg -target /Volumes/Install\ macOS\ Catalina
> mount: / failed with 77 (77)t_apfs: volume could not be mounted: Operation not permitted
> installer: Error - Cannot install without ensuring the file system is read-write
> ```

Sir you are the cause of the error. "Cannot install without ensuring the file system is read-write" means you did not accepted a pre install action that require you to interact into the GUI directly.

bpresles, 2019-10-19:
> Hi, what happens if you write your administrative account name and your administrative account password?

> Sir you are the cause of the error. "Cannot install without ensuring the file system is read-write" means you did not accepted a pre install action that require you to interact into the GUI directly.

No, it's the error that is reported AFTER having entered Administrator credentials.

vectorsigma72, 2019-10-19:
> It shows the error shown in the screenshot:
> 
> https://user-images.githubusercontent.com/220623/66786465-b0ee4100-eee0-11e9-96c2-312c2831f27d.png

Can you post the install log? Just after the error go to menu->Window->Install log

vectorsigma72, 2019-10-19:
> > > On Catalina (OS X 10.15), Clover installer can't get read/write access
> > > [image]
> > > [image]
> > > Of course I've SIP completely enabled (0x0) ;). I didn't try without SIP, but I don't want to have to disable it each time I need to update Clover
> > 
> > 
> > Hi, what happens if you write your administrative account name and your administrative account password?
> 
> It shows the error shown in the screenshot:
> 
> https://user-images.githubusercontent.com/220623/66786465-b0ee4100-eee0-11e9-96c2-312c2831f27d.png
> 
> > > Same here.
> > > ```
> > > g3nsvrv@G3NSVRVs-Mac ~ % sudo mount -u -o rw /Volumes/Install\ macOS\ Catalina 
> > > g3nsvrv@G3NSVRVs-Mac ~ % sudo installer -pkg ./Downloads/Clover_v2.5k_r5096.pkg -target /Volumes/Install\ macOS\ Catalina
> > > mount: / failed with 77 (77)t_apfs: volume could not be mounted: Operation not permitted
> > > installer: Error - Cannot install without ensuring the file system is read-write
> > > ```
> > 
> > 
> > Sir you are the cause of the error. "Cannot install without ensuring the file system is read-write" means you did not accepted a pre install action that require you to interact into the GUI directly.
> 
> No, it's the error that is reported AFTER having entered Administrator credentials.
> 
> In a nutshell, the installer is unable to get read/write access even if you enter the appropriate credentials of an Administrator account in the dialog shown in the first screenshot: https://user-images.githubusercontent.com/220623/66786460-adf35080-eee0-11e9-9de0-f2352b0d3dc0.png

Sure the install will fail if you run the package from the command line. no way

bpresles, 2019-10-19:
> > It shows the error shown in the screenshot:
> > https://user-images.githubusercontent.com/220623/66786465-b0ee4100-eee0-11e9-96c2-312c2831f27d.png
> 
> Can you post the install log? Just after the error go to menu->Window->Install log

Oct 19 11:21:37 iMac Installer[58510]: Package Authoring Error: <background> has an unsupported MIME type: image/data
Oct 19 11:21:37 iMac Installer[58510]: Package Authoring Error: <background-darkAqua> has an unsupported MIME type: image/data
Oct 19 11:21:37 iMac Installer[58510]: Package Authoring Error: <background_scaling> has an unsupported MIME type: X-NSObject/NSNumber
Oct 19 11:21:37 iMac Installer[58510]: Package Authoring Error: <background_alignment> has an unsupported MIME type: X-NSObject/NSNumber
Oct 19 11:21:37 iMac Installer[58510]: Package Authoring Error: <layout-direction> has an unsupported MIME type: X-NSObject/NSNumber
Oct 19 11:21:38 iMac Installer[58510]: Could not load resource license: (null)
Oct 19 11:21:58 iMac Installer[58510]: Installation checks failed.
Oct 19 11:21:58 iMac Installer[58510]: Installation check failure.  . Installation impossible sans passer le système de fichier en lecture/écriture.
Oct 19 11:22:01 iMac Installer[58510]: Package Authoring Error: Infinite loop between <choice> attributes and mixed state aborted.

G3NSVRV, 2019-10-19:
> Sir you are the cause of the error. "Cannot install without ensuring the file system is read-write" means you did not accepted a pre install action that require you to interact into the GUI directly.

Bruh, Do you think that I didn’t try it through the UI? Right click open, put admin credentials, get same error on EULA screen?

vectorsigma72, 2019-10-19:
no surprise if it fails from the command line because there is a check that abort the installation if you did not accept it:

`function checkFileSystemIsWritable()
	{
    var writable = true;
    var cmd = '[[ yes == $(diskutil info / | grep -i \'read-only volume:\' | awk \'{print $NF}\' | tr \'[:upper:]\' \'[:lower:]\') ]]'
    r = system.run('/bin/bash', '-c', cmd);
    system.log('checkFileSystemIsWritable diskutil result = ' + r);

    writable = (r != 0);
    
    if (writable) {
      system.log('checkFileSystemIsWritable: file system is writable: Passed');
    } else {
      system.log('checkFileSystemIsWritable: root file system is read only, try to make it RW...');
      var cmd2 = '/usr/bin/osascript -e\''
      cmd2 += 'do shell script "mount -uw /'
      cmd2 += '" with prompt '
      cmd2 += '"' + system.localizedString('MakingRootRW') + '"'
      cmd2 += ' with administrator privileges\''
      
      try
      {
        r2 = system.run('/bin/bash', '-c', cmd2);
        writable = (r2 == 0);
        if (writable) {
          system.log('mount -uw / succeded.');
        }
      } catch (e) {
        system.log('mount -uw / returned with error: ' + e);
        writable = false;
      }
    
      if (writable == false) {
        my.result.message = system.localizedString('GoToFailRootRW');
        my.result.type = 'Fatal';
      }
    }

		return writable;
	}`
so there is no way you can skip it (i.e. even if the file system was already writable). A failure after giving a correct administrative account name/password can be a bug instead. But the try-catch block should  print 'mount -uw / succeded.' when succeded or 

> 'mount -uw / returned with error: ' + e
when it fails...
but there is no trace of them just as you refuse to go ahead or anyway some how the installation-check is refused, if not by you, is refused by the system. So the question is: are you running from an Administrative account? Why the user name did not appear in your screen shot? Did you remove it for your privacy?

vectorsigma72, 2019-10-19:
> Bruh, Do you think that I didn’t try it through the UI? Right click open, put admin credentials, get same error on EULA screen?

do the questions bother you? Any way from your log, this time I can see an comprehensive and illuminating
`mount: / failed with 77 (77)t_apfs: volume could not be mounted: Operation not permitted`

bpresles, 2019-10-19:
> Why the user name did not appear in your screen shot? Did you remove it for your privacy?

Yes

bpresles, 2019-10-19:
> So the question is: are you running from an Administrative

Yes

vectorsigma72, 2019-10-19:
> Yes

Ok thanks, that is clear. Usually administrative account show the name while guests or non administrative accounts require to fill both user name and password, why I asked.
Have you notice a line in your log that starts with:
`mount: / failed with`
as for @G3NSVRV ? 

some other questions:
Is that happening on the root file system or on another disk only, like an external one? Happens on both cases or one of the two works?
Is that happening because you are targeting a volume containing a Catalina installer or another Catalina partition?

bpresles, 2019-10-21:
> > Yes
> 
> Ok thanks, that is clear. Usually administrative account show the name while guests or non administrative accounts require to fill both user name and password, why I asked.
> Have you notice a line in your log that starts with:
> `mount: / failed with`
> as for @G3NSVRV ?

I gave you the whole log from the Installer, so I guess no ;)

> 
> some other questions:
> Is that happening on the root file system or on another disk only, like an external one? Happens on both cases or one of the two works?

I'm running the installer from the root filesystem.

> Is that happening because you are targeting a volume containing a Catalina installer or another Catalina partition running from a Catalina installation?

I'm trying to install Clover while booted in Catalina from my main SSD (with previous clover). There is no Catalina installation partition or app (nothing in /Applications).

riverLethe9, 2019-10-22:
I found this article on Apple's website that may be useful.
[About the read-only system volume in macOS Catalina](https://support.apple.com/en-us/HT210650)

gavineadie, 2019-11-01:
Returning to this issue .. the bottom line is that the Clover installer seems dysfunctional when used from a Mac running Catalina in that you cannot reach the "click 'Continue' twice" step of the Clover instructions because the installer aborts before offering that option.  I presume, since I don't know the intervals of the process, that the Clover installer always checks if it can write to the root volume, even if the user has no intention of doing that (the more common target of the install being a USB stick), when it finds it can't do that it aborts.

Apple's article, referenced above, points to the reason for the failure, which is nice to know but doesn't lead to any resolution.  So, my questions are:

.. is this impacting all Catalina users or only some ?

.. can this 'writable root volume' check be avoided ?
  .. if not, is Clover useless in macOS 10.15+ ?
  .. if so, is a Clover issue to be addressed ?

My personal solution is to use a pre-Catalina Mac when needing to use the Clover installer to set up USB sticks, hardly an optimal solution.

gavineadie, 2019-11-01:
I just pulled the Clover repo and as @vectorsigma72 notes, the key to this lies in the Installer "Distribution" script in the function `checkFileSystemIsWritable()` where a `mount -uw /` conditions the success of the installation.

Seems like the failure side of that test, instead of making efforts to make root writeable (which can't be successful in Catalina), should work a little harder and look for other writable (bootable?) volumes.  Sadly, the last time I messed with Installer scripts was in the Mac System 7 days, so maybe someone with modern Installer skills could do that (or something better).

Papavero3535, 2019-11-01:
Same issue here. Any fix in sight??

gavineadie, 2019-11-01:
A fast, but unsatisfactory, fix would be to set `my.result.type` to `Warning` instead of `Fatal` when the root is not writable .. a one word fix in the Installer script!  And, yes, I've tested it!!

Papavero3535, 2019-11-01:
> A fast, but unsatisfactory, fix would be to set `my.result.type` to `Warning` instead of `Fatal` when the root is not writable .. a one word fix in the Installer script! And, yes, I've tested it!!

Thanks gavineadie -  Unfortunately I am not very familiar with how to edit the script. any chance you could either are your new package or share some knowledge? thanks!

gavineadie, 2019-11-01:
As a stranger to this product, I feel reluctant to step in.  I don't recommend patching the file in the installer, if it's even possible, because it'll get changed back on the next upgrade.  I built the entire installer from scratch but I'm a programmer and familiar with the tools and processes required.  I expect this will be fixed soon in a better way than my patch.

Papavero3535, 2019-11-01:
> As a stranger to this product, I feel reluctant to step in. I don't recommend patching the file in the installer, if it's even possible, because it'll get changed back on the next upgrade. I built [...]

Definitely understand. Appreciate the extra info though!

Erictoby, 2019-11-05:
Maybe, it's time to move /EFI-Backup folder to somewhere, such as /usr/local/EFI-Backup or ~/EFI-Backup...

juliendonck, 2019-11-05:
Have the same issue. Any chance for an update?

gavineadie, 2019-11-05:
This is not an issue of where things are put, it is merely checking that "/" can be mounted writable.  If it's not (and it is not in Catalina), the Installer immediately aborts, making the issue of where anything gets put beside the point.

Erictoby, 2019-11-05:
@gavineadie  The purpose of checking "/" is writable is to create /EFI-Backup.

gavineadie, 2019-11-05:
.. but, if your intention is to create "/Volumes/USB/EFI_Backup", and not touch "/" at all (a not-uncommon usage), the Installer script always aborts before you get to chose the install volume because it prematurely checks the success of `mount -uw /`.

vectorsigma72, 2019-11-06:
> @gavineadie The purpose of checking "/" is writable is to create /EFI-Backup.

No. RC Scripts needs that, the Clover preference file needs that as well, and so on. The system knows this and wont start the package because potentially can install in /.
Since the work around was to make / writable before any check on any volume and as you cannot do that on any volume that is not "/" ... the package fails.
The fix is just not using the package: [Clover.app](https://www.insanelymac.com/forum/topic/341047-cloverapp-testing/) is the way to go. Under testing.

closed 2019-11-21 by SergeySlice

mnsrit, 2019-11-26:
Hmm that's odd. It seems only confined to the latest version of Clover 5099. I was able to install Clover version 5089 on USB stick with Catalina installer in it. I was able to install it but got a black screen through.

douira, 2019-12-12:
A similar thing happens when I use Clover.app to install to a removable USB device. It says this:
```
Installation begin..
0:142: 
execution error: My Path = /Applications/Hackintosh/Clover.app/Contents/MacOS/Cloverhelper
Target volume: /Volumes/EFI 1
Error: target volume "/Volumes/EFI 1" is not writable. (1)

Installation failed.
```

bchernicoff, 2019-12-15:
I cloned the repo, did a checkout on the 5100 tag, and hard coded checkFileSystemIsWritable to return true. Built the package and ran it from Catalina and was able to install Clover on a USB drive without issue. They need to update the installer to check the permissions on the selected install volume not the root volume of the system running the installer.

bmukheja, 2020-01-04:
What steps did you follow to built the package? @bchernicoff

bchernicoff, 2020-01-04:
@bmukheja With the latest xcode and command line tools installed, I just ran the buildme script which will present a menu asking what to build. I believe I chose Build Package but maybe I chose Build All. Sorry, I don't remember exactly.

fallenprogrammr, 2020-01-15:
Circumvented it by -
Modifying CloverPackage/package/Distribution file
Change functioncheckFileSystemIsWritable in the file to:

`function checkFileSystemIsWritable()
	{
                var writable = true;
		return writable;
	}`
Run buildme from command line
Select option 3 (make pkg)
Run newly created package from CloverPackage/sym/

norda72, 2020-01-26:
Can Apple have blocked Clover in Catalina? To protect their system?

gavineadie, 2020-01-26:
No .. this is a feature of Catalina.  It's not only Clover that's impacted .. Finder has the same restriction.

gomsoup, 2020-02-10:
I fixed this problem with manual down-below

1. restart your mac
2. press command + r until appear Apple's logo
3. go to menubar-utility-terminal
4. type this command
'''csrutil disable'''

profit

rafaelmx, 2020-02-11:
Gomsoup's solution worked for me! Thanks!

bchernicoff, 2020-02-11:
> Gomsoup's solution worked for me! Thanks!

Yeah but now you've disabled an important security feature of your Mac. I would re-enable now that you've made the USB drive.

roddy20, 2020-02-11:
SIP is disabled by Clover as default
<key>CsrActiveConfig</key>
		<string>0x3E7</string>

bchernicoff, 2020-02-12:
> "Of course I've SIP completely enabled (0x0) ;)."
> 
> SIP is disabled by Clover as default
> CsrActiveConfig
> 0x3E7
> 
> if you enabled SIP you cannot install some packages

Sure. What's that got to do with the Apple Mac he was disabling SIP on to run the Clover installer?

roddy20, 2020-02-12:
> Sure. What's that got to do with the Apple Mac he was disabling SIP on to run the Clover installer?

It is not related with Clover and it's development
Mac doesnt require Clover to boot
If you want use your Mac in any unusual and unsupported way you may need to disable SIP
Do you want to change the system and protect it from changes, simultaneously?

roddy20, 2020-02-12:
and the second thing, no need to use installer to update Clover
Just copy BOOTx54.efi and/or CLOVERx64,efi

RedxTar, 2020-05-28:
Im having the same issue on Catalina. Is there a workaround that doesn't involve disabling the SIP just to install it on the USB?

[image]
[image]
[image]

douira, 2020-05-28:
Not a proper solution, but you can partially disable parts of SIP in the Options in the boot screen without disabling SIP altogether. You only need to enable "Allow Unrestricted FS"

ZinkDifferent, 2020-06-01:
This still doesn't work. So basically Clover simply can't be installed on Catalina?

guizy67, 2020-09-09:
Clover does work with Catalina, just make sure your `config.plist` is correct.

[image]

I was having the same issue until I corrected these values (they were initially blanc)