CloverHackyColor/CloverBootloader #7: Extremely slow FileVault boot on Catalina beta 11 / Final 19A583

bpresles, 2019-10-15:
When booting a FileVault encrypted apfs volume it sits for a very long time (1-2 min) at an apple logo with black background before showing the FileVault password entry screen. This is a regression in Catalina beta 11 / final as it was working instantaneously on all versions of Mojave and Catalina beta 9 and prior. Verbose mode doesn't work at this stage in booting.

This happens with current Clover latest release build: 5096 and previous (verified on 5070 and onward). 

Originally reported both on Reddit https://www.reddit.com/r/hackintosh/comments/d1m24i/extremely_slow_filevault_boot_on_catalina/

And on InsanelyMac
https://www.insanelymac.com/forum/topic/339035-pre-release-macos-catalina/?page=69&tab=comments#comment-2692344
https://www.insanelymac.com/forum/topic/339035-pre-release-macos-catalina/?page=69&tab=comments#comment-2692364

blueberry6401, 2019-10-24:
Hi, any working on this?

thenickdude, 2019-11-19:
Possible solution here, looks like some EFI drivers need to be swapped out:

https://www.tonymacx86.com/threads/filevault-2-preloader-takes-long-time.285306/#post-2021475

blueberry6401, 2019-11-19:
> Possible solution here, looks like some EFI drivers need to be swapped out:
> 
> https://www.tonymacx86.com/threads/filevault-2-preloader-takes-long-time.285306/#post-2021475

Thank, that fixes the issue 🍡

thenickdude, 2019-11-19:
My test machine finished locking its drive, and now I can confirm the fix works for me too!

bpresles, 2019-11-19:
Fixing the issue using AppleSupportPkg EFI drivers can only be a temporary solution as these drivers are made for OpenCore normally (and have been even integrated in it since).

The drivers included with Clover have to be fixed themselves.

matis-schotte, 2019-12-18:
Any update on this?

Is it fine to update clover using the pkg? I feel hesitant since I have a modified install now.
Or just udpate the clover.bin file?

real-cereal-boy, 2020-01-16:
If it doesn’t work, you might want to turn off csrutil, Catalina verifies the disk so that is probably why
** Use at own risk **

arrmo, 2020-03-29:
Hi,

The driver swap works here as well (temporarily of course) - at least to get through the very long wait. But when the password screen comes up .. where I need to enter the password is just a square of "missing" graphics / text (pic below). Is there a way to address this?

[image]

Thanks!

thenickdude, 2020-03-29:
Same thing happens to me, I've just been typing the password in blind.

arrmo, 2020-03-29:
Hmmm ... assuming it's a driver related thing. Have you tried OC, just to get to the bottom of it?

Thanks!

bpresles, 2020-05-08:
@SergeySlice @asava @vectorsigma72 Any news on that issue ? For now we can workaounrd it by using old EFI drivers from OpenCore, but it'd be nice to have it fixed in Clover's included drivers also :)

asava, 2020-05-10:
I am not sure, I don't use FileValut so I cannot test myself, which makes it more difficult.
Are you sure you need to replace all mentioned drivers there? Or it's just a matter of one of them (I would guess InputFix)? If you point me to something more specific, maybe I can figure out what it is.

SergeySlice, 2020-05-11:
I see any log to speak about.
May be users install VirtualSMC but using SMCHelper which is not compatible.

bpresles, 2020-05-11:
@SergeySlice I personnally use FakeSMC and SMCHelper.efi. I also tried VIrtualSMC with VirtualSMC.efi and no changes are observed.

@asava For the log, at least there are nothing printed, even with "-v" boot flag, on screen. I'll try with Clover EFI debug log activated.

asava, 2020-05-11:
Also please specify exactly which drivers are the minimum you need to replace to have it working. I assume it would be a single driver.

bpresles, 2020-05-11:
Clover debug log when using Clover's provided FileVault2 drivers:

- AppleImageCodec.efi
- AppleUITheme.efi
- AppleKeyAggregator.efi
- FirmwareVolume.efi
- HashServiceFix.efi

[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4612056/debug.log)

asava, 2020-05-11:
Maybe I didn't explain clear enough.
That thread you linked to, talks about replacing many drivers.
We need to know exactly, which driver you are replacing that fixes it.

bpresles, 2020-05-11:
I was answering @SergeySlice comment, not yours

I’m working on yours but because of this bug it takes more than 2 full minutes to boot each time I try! So please be patient!

bpresles, 2020-05-11:
Ok, I tried the following:

- Replacing AppleKeyAggregator.efi by AppleGenericInput.efi from OpenCore => Not booting at all (not even the Apple logo)

... to be continued

arrmo, 2020-05-11:
BTW, with this driver swap - do you still get the big square block I mentioned above?

Thanks!

bpresles, 2020-05-12:
> BTW, with this driver swap - do you still get the big square block I mentioned above?
> 
> Thanks!

I never had this bug on my setup.

arrmo, 2020-05-12:
FYI, I tried your fix (files) above - it works, thanks! But I still have the block issue above. What is your host OS? That may be part of it - I'm on Linux, not sure what you are using.

BTW, for OC, this fix works, but not sure how to set this same thing in Clover?
https://github.com/acidanthera/bugtracker/issues/751#issuecomment-624987291

Thanks again.

bpresles, 2020-05-12:
I don’t have host OS. I’m running macOS natively, not on a VM.

asava, 2020-05-12:
> **Conclusion:** The driver causing the issue is **AppleImageCodec.efi**. Also note that AppleUISupport from OpenCore contains features from all following Clover's drivers AppleUITheme.efi, AppleImageCodec.efi, HashServiceFix.efi and FirmwareVolume.efi (as you can see in source code link below) so when replacing using AppleUISupport.efi instead of AppleImageCodec.efi, all these drivers can also be removed.

Kind of a shot in the dark, but I do see one possible problem. Please try this driver:
[AppleImageCodec.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4618336/AppleImageCodec.zip)

bpresles, 2020-05-12:
Thanks I’ll try it right now.

Btw, you really should have a test computer with FileVault 2 activated to be able to reproduce FV2 issues in the future to avoid to fix with “shots in the dark”, as you said ;)

bpresles, 2020-05-12:
Ok, I’ve tried your attached driver. It doesn’t fix the issue. The FV2 UI is still taking ages to appear with this driver.

I’ll provide the log once I finish booting... But still I think that you should have a test machine with FV2 to reproduce issues related to FV2.

asava, 2020-05-12:
Well, I tried to make some changes that looked it could affect it. But
maybe it's better to leave the issue to someone who has FileVault, as I
definitely don't plan to activate it.

SergeySlice, 2020-05-18:
There is an efi driver that can help us to understand the issue. Place it in drivers and after boot you may find a log
/efi/clover/misc/eficalls.log
[DumpUefiCalls.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4645283/DumpUefiCalls.efi.zip)

jief666, 2020-06-01:
I have a problem with Catalina/FV. it stays at white Apple on black background.
The problem comes from Preboot volume/[GUID]/System/Library/Caches/com.apple.corestorage/EFILoginLocalizations/loginui.efires.
With High sierra, it's one MB. With Catalina (testing 10.15.4) it's 56MB.

So I copy the high sierra version over, and, except that I have a blur high sierra image, it's working fine.

SergeySlice, 2020-06-06:
So huge image == long boot?
If not load the image with OpenCore drivers then fast boot without password prompt.

jief666, 2020-06-21:
After (a lot of) further investigations and some luck, I found what was causing it for me : I just removed AppleImageCodec-64.efi from drivers folder.
I don't know what it's supposed to do, but boot Mojave (I think I also tried high sierra) fine without it.
I didn't try Catalina, who have an even bigger loginui.efires (56MB on Catalina, 26MB on Mojave).

jief666, 2020-06-21:
Forgot to say, for Mojave boot, the pause time on the apple logo before being able to enter password went from 35 seconds to 4 seconds.

MysticalOS, 2020-07-17:
This issue can be solved much more easily now that clover has plimented OC quirks and core files. literally all the filevault 2 stuff was merged into those after they deprecated the stand alone efi. Clover should clean house on all the obsolete EFI drivers at this point and this issue wold be solved

closed 2020-09-19 by SergeySlice