CloverHackyColor/CloverBootloader #9: APFS boot from NVMe not possible, Clover does not see NVMe

joergmaertin, 2019-10-18:
I installed a new NVMe Crucial 1TB via PCIe Adapter. Clover 2.5k_r5093 sees this disk only if formatted HFS+. I tried to clone a Catalina Installation to the NVMe, no success. I tried to install Catalina fresh, Clover sees the disk after the first step, the installation continues, on last reboot Clover does not see the NVMe again. I tried to install Mojave fresh, also the APFS isn't available in Clover. macOS Catalina sees the disk and it is usable, but not bootable if formatted APFS. Is there a solution to this problem?
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/3742312/preboot.log)

[config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/3742315/config.plist.txt)

SergeySlice, 2019-10-21:
Clover know NVMe and found partitions on it.
See other problems here
`9:002  0:000  - [28]: Volume: PciRoot(0x0)\Pci(0x1,0x1)\Pci(0x0,0x0)\NVMe(0x1,00-00-00-00-00-00-00-00)\HD(1,GPT,A0FE4399-A029-4D23-8767-2E6494A719CC,0x28,0x64000)
9:002  0:000          Result of bootcode detection: bootable unknown (legacy)
9:005  0:002  - [29]: Volume: PciRoot(0x0)\Pci(0x1,0x1)\Pci(0x0,0x0)\NVMe(0x1,00-00-00-00-00-00-00-00)\HD(2,GPT,7DB610CB-B00F-4434-8B9E-EE46A046160E,0x64028,0x746A2D60)
9:005  0:000          Result of bootcode detection: bootable unknown (legacy)`

SergeySlice, 2019-12-04:
abandoned

closed 2019-12-04 by SergeySlice