CloverHackyColor/CloverBootloader #11: Clover USB 3.0 port not working (still waiting for root device)

AlbertoHugonin, 2019-10-21:
On my two computers usb 3.0 cannot be used as a method to install osx, each version newer than 2.4k_r4813 breaks usb 3.0 (still waiting for root device). If you can start the installer using a usb 2.0 you can then install usbinjectall and the 3.0 ports will work. On my laptop however I don’t have usb 2.0 so i can’t proceed. I am writing this because I think that this is a clover issue because if I use 4813 it works but if I use another newer version it does not.
[clover.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3749472/clover.zip)

SergeySlice, 2019-10-21:
There are no changes in Clover since 4813 that may influence on USB 3.0.
Provide please preboot.log for both cases.

AlbertoHugonin, 2019-10-21:
@SergeySlice 
[v2.4k_r4813.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3752158/v2.4k_r4813.zip)
[v2.4k r4895.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3752160/v2.4k.r4895.zip)
[v2.5k r5097.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3752164/v2.5k.r5097.zip)

Only in r4813 usb 3.0 works and I can boot the installer. With any other version newer I am stuck at still waiting for root device.

AlbertoHugonin, 2019-10-22:
I tried to use this USB port limit patch for 10.14.6 and now I can boot Mojave: [https://www.tonymacx86.com/threads/usb-port-limit-patch-for-14-1-14-2-14-3-14-5-14-6.268893/](url)
My question is why If I use 2.4k_r4813 I can boot without this usb patch and on 4895 and newer not?

MysticalOS, 2019-10-22:
On 10.14.x you should be fixing your USB ports to have a proper port injector and remove all dummy ports, that way you don't need port limit patches anymore. USB ports just always work. 

Basically involves temp using limit patch, usinga tool to show USB ports to identify them like hackintool, plugging and unplugging USB 3.0 and 2.0 flash drives ideally so IDs for both modes on each port get identified, then saving port map and using an injector like USBInjectAll to inject ONLY that port map at boot instead of everything (including dummy ports) it finds in ACPI. Then you'll stop having port limit issues and your USB bus will even be far more accurate since it'll be mapped correctly.

AlbertoHugonin, 2019-10-23:
@MysticalOS I will do that but for now my USB ports just works.
My only concern is why my clover setup is acting so strangely.
After installing Mojave and usbinejctall I don’t need the USB port limit patch anymore, the usb just works only with the kext installed
If I don’t need the USB port limit patch and I have usbinjectall inside kext/other folder why I have to use that usb patch to booting the installer?
And also why this is happening only after 2.4k_r4813?
Is not a problem anymore because I now have Mojave installed correctly this is just for my curiosity.

closed 2019-12-30 by SergeySlice