CloverHackyColor/CloverBootloader #13: Black screen with 5700 XT Catalina Beta 2(19B77a) with WhateverGreen

errorexists, 2019-10-23:
just tried the new Weg an still a black screen when booting on clover for second stage also tried boot args (agdpmod=pikera,vit9696) using either DP an HDMI ports with same results

telepati, 2019-10-23:
Also, there is a bug in WEG @SergeySlice warned me. can't boot within Safe Mode; I attached verbose screen on official WEG thread;
https://www.insanelymac.com/forum/topic/325987-whatevergreen-support-topic/?do=findComment&comment=2694753

kleuter, 2019-10-25:
5700 XT is not yet supported by macOS

errorexists, 2019-10-25:
yes the 5700 XT is suported in mac os as i said in the head line im on catalina beta 2 an as stated in the release notes that it is supported

MysticalOS, 2019-10-25:
It actually is by 10.15.1 if you did research, but it's support is not complete by WEG yet. WEG has some 10.15.1 checked into their branch but not pushed as a release yet. there is an ongoing ticket there.

I imagine clover probably needs to update it's hardware IDs as well?

kleuter, 2019-10-25:
Oops, thanks for the info

SergeySlice, 2019-11-16:
> It actually is by 10.15.1 if you did research, but it's support is not complete by WEG yet. WEG has some 10.15.1 checked into their branch but not pushed as a release yet. there is an ongoing ticket there.
> 
> I imagine clover probably needs to update it's hardware IDs as well?

Already done
`{ 0x731F,  CHIP_FAMILY_NAVI10, "AMD Radeon RX5700",        kNull },`

SergeySlice, 2019-11-16:
There is no issue for Clover
- using WEG you have to report to WEG
- without WEG you have to set properties for the card to be recognised by OS Catalina. It is not default for Clover.

closed 2019-11-16 by SergeySlice