CloverHackyColor/CloverBootloader #15: Fix version from tag logic

Dids, 2019-10-24:
There's an issue with [this](https://github.com/CloverHackyColor/CloverBootloader/commit/ad2972de78132fd9839cfe34b26519832c992918) (and [this](https://github.com/CloverHackyColor/CloverBootloader/commit/807a11a1b71a2c3573d46e9cfabd8c69e62647f4)), where it'll always use the newest tag, even if you're currently not on the latest tag.

For example, say you've checked out `tags/5096`. With tag `5097` being the latest, Clover would still be built with `5097` set as the revision/version.

Here's the current command and its output:
```
❯ git describe --tags $(git rev-list --tags --max-count=1)
5097
```

Now here's one potential way you could fix this:
```
❯ git describe --tags --abbrev=0                          
5096
```

This way it would always use the current tag, _or_ the latest one up to the current commit, but never a newer one.

SergeySlice, 2019-10-28:
Good point!

SergeySlice, 2019-10-29:
Accepted in 5098

closed 2019-10-29 by SergeySlice