CloverHackyColor/CloverBootloader #16: Shift key doesn't work in Clover

JingleDjango, 2019-10-25:
Was previously using ver. 5045 and had this issue. Just updated to current stable release (5097) and the issue persists.

In Clover, for example when trying to manually add boot flags in the Options menu, the Shift key does nothing. I've tried two wired USB keyboards: Matias QuietKey Pro (Mac) and a Logitech K120 (Win). 

Searching around I've seen similar reports on various forums but they date back several years and none look to be resolved.

SergeySlice, 2019-10-31:
It is BIOS feature, not resolvable.

JingleDjango, 2019-11-02:
Thanks for your reply.

Is the problem specific to certain BIOSs? 
Would it be possible to reassign another key to that character?

chrisbutler, 2019-11-05:
@SergeySlice if this is true, why does it work fine in some versions of clover?

JingleDjango, 2019-11-25:
That was my question next question as well.

SergeySlice, 2019-12-30:
> @SergeySlice if this is true, why does it work fine in some versions of clover?

If it doesn't work with this BIOS then it will not work with other Clover version.

closed 2020-02-11 by SergeySlice