CloverHackyColor/CloverBootloader #18: Clover update issue in Catalina

Hillperson, 2019-10-26:
I've updated the clover to 5096 before, and updated my mojave to catalina. Everything is good and no problem. I have updated to supplement version now 19A602.
I just got a notice that clover 5097 is released and it suggested me to update. So I download it from the Github.
When I try to install clover 5097, it said the installer is not compatible with this MacOS, and when I continue, it finally failed. I can still reboot from clover, but the clover control panel disappeared from the system preference. I also tried to install the clover 5096 again, the same error.

I guess reason is about the new system file protection in Catalina, but I don't know how to solve it and update my clover in future.

Any solution to this? Thanks

reginastracii, 2019-10-26:
read here : [https://www.tonymacx86.com/threads/macos-10-15-catalina-how-to-mount-root-partition-read-write-for-post-installation.283749/](url)

Hillperson, 2019-10-27:
> read here : `https://www.tonymacx86.com/threads/macos-10-15-catalina-how-to-mount-root-partition-read-write-for-post-installation.283749/`

Thanks for the help, but I tried that command, and make sure there was no error during the command, but the installation is still failed.

I posted the issue in Tonymacx86 as well, I put 2 pictures for my problem. And I double check the real clover version on the EFI partition with restart, it is still 5097 not 5096. Which means the install failed, and no change in EFI, and no control panel in the system preference.

wuyuehyang, 2019-10-27:
Same issue here. My Catalina was upgraded from Mojave too.

barad1tos, 2019-10-27:
+1. 

Same issue and the same upgrade to Catalina from Mojave.

Hillperson, 2019-10-29:
use the new 5098 version. Problem solved.

closed 2019-10-29 by Hillperson

wuyuehyang, 2019-10-30:
[image]

r5098 doesn't work either. Root partition is already mounted as r/w.

My Catalina is on 10.15 (19A583).

CreativeMK, 2019-11-02:
E3-1230 v3 1150 Hasewell CPU, MSI ZH87 motherboard, HD7950 graphics card.

I have the same problem too. And I tried both r5070 and r5097. They all stucked at restart. 
With r5097, it just restart again and again.

<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>#DisableDrivers</key>
	<array>
		<string>CsmVideoDxe</string>
		<string>VBoxExt4</string>
	</array>
	<key>ACPI</key>
	<dict>
		<key>#PatchAPIC</key>
		<false/>
		<key>#ResetAddress</key>
		<string>0x64</string>
		<key>#ResetValue</key>
		<string>0xFE</string>
		<key>#SortedOrder</key>
		<array>
			<string>SSDT-3.aml</string>
			<string>SSDT-1.aml</string>
			<string>SSDT-2.aml</string>
		</array>
		<key>#smartUPS</key>
		<false/>
		<key>AutoMerge</key>
		<false/>
		<key>DSDT</key>
		<dict>
			<key>#DropOEM_DSM</key>
			<dict>
				<key>ATI</key>
				<true/>
				<key>Firewire</key>
				<true/>
				<key>HDA</key>
				<true/>
				<key>HDMI</key>
				<true/>
				<key>IDE</key>
				<true/>
				<key>IntelGFX</key>
				<true/>
				<key>LAN</key>
				<true/>
				<key>LPC</key>
				<false/>
				<key>NVidia</key>
				<true/>
				<key>SATA</key>
				<true/>
				<key>SmBUS</key>
				<false/>
				<key>USB</key>
				<true/>
				<key>WIFI</key>
				<true/>
			</dict>
			<key>#Rtc8Allowed</key>
			<false/>
			<key>#SuspendOverride</key>
			<false/>
			<key>Debug</key>
			<false/>
			<key>DropOEM_DSM</key>
[... 1139 lines cut ...]
		<string>0x01</string>
		<key>ProductName</key>
		<string>iMac14,2</string>
		<key>SerialNumber</key>
		<string>D25P621SF8JC</string>
		<key>Version</key>
		<string>1.0</string>
	</dict>
	<key>SystemParameters</key>
	<dict>
		<key>#BacklightLevel</key>
		<string>0x0501</string>
		<key>#CustomUUID</key>
		<string>511CE201-1000-4000-9999-010203040506</string>
		<key>#NoCaches</key>
		<false/>
		<key>#NvidiaWeb</key>
		<false/>
		<key>InjectKexts</key>
		<string>Yes</string>
		<key>InjectSystemID</key>
		<true/>
	</dict>
</dict>
</plist>

CreativeMK, 2019-11-02:
E3-1230 v3 1150 Hasewell CPU, MSI ZH87 motherboard, HD7950 graphics card.

I have the same problem too. And I tried both r5070 and r5097. They all stucked at restart.
With r5097, it just restart again and again.

Sorry for uploading error. Something went wrong. Can't process file.

But after SMC sucessfully initialized.....
error: invalid CORB size: 00
error: getCapabilities failed
apfs_module_start:1683: load: com.apple.filesystem.apfs, v1412.41......

end.