CloverHackyColor/CloverBootloader #19: 10.15 AppleIntelMCEReporter

zhang-wenchao, 2019-10-26:
https://github.com/acidanthera/bugtracker/issues/424

Clover can solve it?

SergeySlice, 2019-10-28:
There is no problem for Clover.

zhang-wenchao, 2019-10-29:
I use the Clover found the same problem.

zhang-wenchao, 2019-10-29:
My CPU is Intel Xeon Gold 5117，must fackcpuid 0x0306E0 to boot。

zhang-wenchao, 2019-10-29:
And, 10.15 panic
[image]

zhang-wenchao, 2019-10-29:
```
mp_kdp_enter() can't get x86_topo_lock! Debugging anyway! #YOLO
panic(cpu 30 caller 0xffffff80174689ca): Kernel trap at 0xffffff7f99995e5d, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0xffffff83a25b5000, CR3: 0x000000002e681000, CR4: 0x00000000003626e0
RAX: 0x3f0011238c5240ca, RBX: 0x0000000000000000, RCX: 0x000000000000001e, RDX: 0x0000000000000000
RSP: 0xffffff83c3d50e70, RBP: 0xffffff83c3d50eb0, RSI: 0xffffff8068100ac0, RDI: 0x0000000000000002
R8:  0x0000000000000001, R9:  0x0000000000000000, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0xffffff8069585480, R13: 0x0000000000000168, R14: 0xffffff8017e53648, R15: 0xffffff83c380bc60
RFL: 0x0000000000010087, RIP: 0xffffff7f99995e5d, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0xffffff83a25b5000, Error code: 0x0000000000000000, Fault CPU: 0x1e, PL: 3, VF: 0

Backtrace (CPU 30), Frame : Return Address
0xffffff8017160220 : 0xffffff801733f98b 
0xffffff8017160270 : 0xffffff8017476c15 
0xffffff80171602b0 : 0xffffff801746861e 
0xffffff8017160300 : 0xffffff80172e6a40 
0xffffff8017160320 : 0xffffff801733f077 
0xffffff8017160420 : 0xffffff801733f45b 
0xffffff8017160470 : 0xffffff8017ad2d89 
0xffffff80171604e0 : 0xffffff80174689ca 
0xffffff8017160660 : 0xffffff80174686c8 
0xffffff80171606b0 : 0xffffff80172e6a40 
0xffffff80171606d0 : 0xffffff7f99995e5d 
0xffffff83c3d50eb0 : 0xffffff8017471bd3 
0xffffff83c3d50f20 : 0xffffff80174718b2 
0xffffff83c3d50f60 : 0xffffff80174707b3 
0xffffff83c3d50f80 : 0xffffff8017467e1b 
0xffffff83c3d50fd0 : 0xffffff80172e6bed 
0xffffff83c36d3f40 : 0xffffff8017364577 
0xffffff83c36d3f80 : 0xffffff8017364738 
0xffffff83c36d3fa0 : 0xffffff80172e613e 
      Kernel Extensions in backtrace:
         com.apple.driver.AppleIntelMCEReporter(115.0)[AAF2300A-7F8A-38EE-B484-05A9F55D7DB4]@0xffffff7f99990000->0xffffff7f999a2fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[9BA2613C-A129-3EDD-8DC1-2CFF6B62B832]@0xffffff7f97ee1000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[E6E855AD-A5B2-3518-A997-1384B4360708]@0xffffff7f97ef3000

BSD process name corresponding to current thread: kernel_task
Boot args: npci=0x3000 

Mac OS version:
19A603

Kernel version:
Darwin Kernel Version 19.0.0: Wed Sep 25 20:18:50 PDT 2019; root:xnu-6153.11.26~2/RELEASE_X86_64
Kernel UUID: 70EDD61F-86EE-3E1B-873F-98D909B78160
Kernel slide:     0x0000000017000000
Kernel text base: 0xffffff8017200000
__HIB  text base: 0xffffff8017100000
System model name: MacPro7,1 (Mac-27AD2F918AE68F61)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 36451591373
last loaded kext at 35946227091: >!A!IMCEReporter	115 (addr 0xffffff7f99990000, size 77824)
loaded kexts:
com.intel.kext.intelhaxm	7.5.1
as.vit9696.WhateverGreen	1.3.3
as.vit9696.Lilu	1.3.8
org.slice.SMIMonitor	1.1.0
org.slice.!ICPUMonitor	1.2.3
org.tgwbd.driver.NullCPUPowerManagement	1.0.0d2
org.slice.RadeonMonitor	1.3.4
org.netkas.FakeSMC	3.5.2
>!A!IMCEReporter	115
>!A16X50ACPI	3.2
>ACPI_SMC_PlatformPlugin	1.0.0
@filesystems.autofs	3.0
@filesystems.ntfs	3.14.3
@filesystems.msdosfs	1.10
|SCSITaskUserClient	422.0.2
@filesystems.apfs	1412.11.7
>!AAHCIPort	341.0.2
>!AFileSystemDriver	3.0.1
>!AVirtIO	1.0
@filesystems.hfs.kext	522.0.9
@!AFSCompression.!AFSCompressionTypeDataless	1.0.0d1
@BootCache	40
@!AFSCompression.!AFSCompressionTypeZlib	1.0.0
@private.KextAudit	1.0
>!AACPIButtons	6.1
>!AHPET	1.8
>!ARTC	2.0
>!ASMBIOS	2.1
>!AAPIC	1.7
$!AImage4	1
@nke.applicationfirewall	302
$TMSafetyNet	8
@!ASystemPolicy	2.0.0
>!A!ICPUPowerManagement	222.0.0
|EndpointSecurity	1
>!A16X50Serial	3.2
>IOPlatformPluginLegacy	1.0.0
>IOPlatformPlugin!F	6.0.0d8
@kext.triggers	1.0
|IOAHCIBlock!S	316.0.5
>usb.IOUSBHostHIDDevice	1.2
>usb.!UHub	1.2
>usb.cdc	5.0.0
>usb.networking	5.0.0
>usb.!UHostCompositeDevice	1.2
|IOSCSIMultimediaCommandsDevice	422.0.2
|IOBD!S!F	1.8
|IODVD!S!F	1.8
|IOCD!S!F	1.8
>!AXsanScheme	3
|IONVMe!F	2.1.0
|IOAHCISerialATAPI	268
>usb.!UXHCIPCI	1.2
>usb.!UXHCI	1.2
|IOAHCI!F	290.0.1
>!A!ILpssGspi	3.0.60
|IOSerial!F	11
|IOSurface	269.6
@filesystems.hfs.encodings.kext	1
>usb.!UHostPacketFilter	1.0
|IOUSB!F	900.4.2
>!AEFINVRAM	2.1
>!AEFIRuntime	2.1
|IOHID!F	2.0.0
$quarantine	4
$sandbox	300.0
@kext.!AMatch	1.0.0d1
>DiskImages	493.0.0
>!AFDEKeyStore	28.30
>!AEffaceable!S	1.0
>!AKeyStore	2
>!UTDM	489.11.2
|IOSCSIBlockCommandsDevice	422.0.2
>!ACredentialManager	1.0
>KernelRelayHost	1
>!ASEPManager	1.0.1
>IOSlaveProcessor	1
|IOTimeSync!F	800.14
|IONetworking!F	3.4
|IOUSBMass!SDriver	157.11.2
|IOSCSIArchitectureModel!F	422.0.2
|IO!S!F	2.1
|IOUSBHost!F	1.2
>!UHostMergeProperties	1.2
>usb.!UCommon	1.0
>!ABusPower!C	1.0
|CoreAnalytics!F	1
>!AMobileFileIntegrity	1.0.5
@kext.CoreTrust	1
|IOReport!F	47
>!AACPIPlatform	6.1
>!ASMC	3.1.9
>watchdog	1
|IOPCI!F	2.9
|IOACPI!F	1.4
@kec.pthread	1
@kec.Libm	1
@kec.corecrypto	1.0

```

SergeySlice, 2019-10-29:
It requires one of the Kext patches developed somewhere for AppleIntelMCEReporter.kext.
The Clover has this functionality. Just write Find/Replace strings.
Or not use MacPro as SMBIOS model.

closed 2019-10-29 by SergeySlice

iMSannob, 2019-11-16:
> ```
> mp_kdp_enter() can't get x86_topo_lock! Debugging anyway! #YOLO
> panic(cpu 30 caller 0xffffff80174689ca): Kernel trap at 0xffffff7f99995e5d, type 13=general protection, registers:
> CR0: [...]

I am always wondering how people captured those messages in boot time.
Could you please teach me how to do that?

Thank you.

useafterfree, 2020-04-29:
> I am always wondering how people captured those messages in boot time.
> Could you please teach me how to do that?
>
> Thank you.

they are booting macOS in KVM, I believe