CloverHackyColor/CloverBootloader #20: Document the switch to Github

chrisbutler, 2019-10-27:
I know documentation is a lot to ask for from an open-source project of this scale... but at least a formal notice on the SourceForge site somewhere would be pretty helpful.

When I download a release from a website and all it has is a text file pointing me somewhere else (with no mention on the actual site), I am (and should be) immediately suspicious that something has been compromised.

A little bit of guidance goes a long way...

Sniki, 2020-02-02:
@SergeySlice is there a specific place where the configuration wiki is located now since the switch to Github ?

The one on Sourceforge hasn't been updated lately, last update date was 24-April-2019 therefore i see many changes have happened to Clover.

Just a tip: it would be nice to make something similar like OpenCore Configuration and Differences have, it would be easier for everyone to help/maintain/update it regularly by following up latest changes.

Thanks,
Sniki

closed 2020-04-29 by asava

RandomDSdevel, 2020-05-01:
@asava:  What resolved this issue, allowing you to close it?

asava, 2020-05-01:
We prefer not to have documentation as part of open issues. It was tagged as documentation, and will be taken care of, when we get it to it. Thank you for your concern.