CloverHackyColor/CloverBootloader #22: Latest Clover fails to install with "cannot install without ensuring file system is read-write" on Catalina

douira, 2019-10-28:
I'm on macOS Catalina 10.15 (19A602). I have not updated Clover since I switched to Catalina from Mojave. SIP Protection is on and everything works fine otherwise. I have not had this problem before upgrading. When I open the installer `Clover_v2.5k_r5098.pkg` with right-click and open it goes through these states after interaction:
[image]
[image]
[image]
In the end, it shows the error "cannot install without ensuring the file system is read-write" and then quits.

valinor61, 2019-10-28:
change crsactive config to 0x67

douira, 2019-10-28:
So this is a SIP issue? I presume I can re-enable it after installing it with SIP disabled?

SergeySlice, 2019-10-29:
sudo mount -uw /

Shaneee92, 2019-10-30:
Disable SIP to install pkg. Can enable once installed.

closed 2019-10-31 by SergeySlice

rgaufman, 2019-11-16:
Same issue here, how do I disable SIP?

effektsvk, 2019-11-16:
@rgaufman Boot into Recovey Mode (while your Mac is starting, press and hold Command + R. Open Terminal (Utilities -> Terminal) and type `csrutil disable` and press Enter. You can restart now. After you use Clover, you can enable it with `csrutil enable`.

douira, 2019-11-16:
Another way is to set the CsrActivateConfig value in the RT Variables section of the Clover config to `0x67`. This is best done using CloverConfigurator. I've had problems using `csrutil disable` but that might also have been my mistake. If you can't boot without disabling SIP in order to change the setting, you also find the setting in the menu of the Clover boot screen.

dmcnaugh15, 2020-03-25:
Thank you douira!  Clover Configurator is the way to go!  Super easy.  Here's a screenshot for anyone else who's interested.

[image]

douira, 2020-03-25:
I've also found out that setting only the second bit (0x02) which controls "UnrestrictedFS" also works if you're really only installing Clover. This will not work for other operations, like installing kexts.

blueberry6401, 2020-04-15:
But why clover want system read-write even if I only want to install it on usb?

jonasrk, 2020-05-04:
Yes, can we re-open this issue please? I, too want to install on an USB, so the clover installer should not need full access to my current system, as outline in the post above. I'm currently stuck for this reason, because I'm prohibited from disabling SIP.

mackonsti, 2020-05-21:
I would like to report that **r5118** also shows this error "Cannot install without ensuring the file system is read-write." after entering admin password; not sure what the PKG script inside needs to write to / as I can mount EFI partition without problems, and there are really no other stuff to write to main-partition-root that the Installer can't write to the data-partition-root (e.g. EFI Backups, settings PLIST etc.)

Please fix this in the installer?

tuyen-vuduc, 2020-07-30:
I tried `csrutil disable` but the message is still. 

My MacbookPro is an early 2013, 15" one

@dmcnaugh15 could you advise where is config.plist to edit? 

Thanks.

douira, 2020-07-30:
@tuyen-vuduc if you're using Clover as the bootloader, disabling SIP is done with `CsrActivateConfig` not `csrutil disable`.

TonyC5, 2020-11-15:
Those that are experiencing the CLOVER "cannot install without ensuring file system read-write" problem.  Did you by any chance install OC on this volume and now you're trying to install CLOVER?