CloverHackyColor/CloverBootloader #24: Dont install NVRAM-Emu with AptioMemoryFix

t14462, 2019-10-29:
Do not install NVRAM-Emu things when user select AptioMemoryFix.efi in installer

t14462, 2019-10-31:
AptioMemoryFix.efi — this driver provide native NVRAM

So, without these files:
/Volumes/EFI/EFI/CLOVER/drivers64UEFI/EmuVariableUefi-64.efi
/Volumes/EFI/nvram.plist
/etc/rc.clover.lib
/etc/rc.boot.d/10.save_and_rotate_boot_log.local
/etc/rc.boot.d/20.mount_ESP.local
/etc/rc.boot.d/70.disable_sleep_proxy_client.local.disabled
/etc/rc.shutdown.d/80.save_nvram_plist.local
boot and power-off 1-2sec faster

Who need an emulator, when native NVRAM works fine?

Sher1ocks, 2019-11-01:
Some board still need RC script with AptioMemoryFix

vectorsigma72, 2019-11-06:
> Who need an emulator, when native NVRAM works fine?

The one who want a boot failure.

SergeySlice, 2019-11-16:
Sometimes need sometimes no.

closed 2019-11-16 by SergeySlice