CloverHackyColor/CloverBootloader #26: v2.5k-5098 installer error on macOS 10.15.1

augiem, 2019-11-03:
The latest release will not install on macOS 10.15.1, failing with the error "An unexpected error occurred while moving files to the final destination."

Log:
https://pastebin.com/LsSq9wnR

Excerpt:
Nov  2 15:19:20 #####s-iMac shove[1446]: [source=file] failed _RelinkFile(/Library/InstallerSandboxes/.PKInstallSandboxManager/903C5760-622B-45B8-A2F1-5EAB219523B4.activeSandbox/Root/EFIROOTDIR/EFI/BOOT/BOOTX64.efi, /EFI/BOOT/BOOTX64.efi): Directory not empty

I've tried:
Disabling SIP
running sudo mount -uw /

benneq, 2019-11-06:
I just installed version 5098 successfully on macOS 10.15.1
I did **not** disable SIP or anything else. Just opened the Clover installer. macOS then ask me if I want to allow the installer to write to the protected file system, clicked yes, and everything worked without problems.

MysticalOS, 2019-11-06:
I wonder if this has to do with relocation execution (running installer from where it was downloaded vs downloading it, moving it, then running it).

macOS will do additional relocation sandboxing based on whether or not the file was moved after download. It's even worse if you moved it after running it, then run it again. OS will know it's not where it was first time it was run and imposes even more scrutiny on it.

then of course there is the obvious question, was it downloaded and run on an admin account. i can't imagine it'd work at all if it wasn't.

benneq, 2019-11-06:
In my case I didn't move it. Just started it from my Downloads folder. And I only have 1 account on my machine. Though I guess it's the admin account.

augiem, 2019-11-06:
Thanks for the replies!

> I just installed version 5098 successfully on macOS 10.15.1
> I did **not** disable SIP or anything else. Just opened the Clover installer. macOS then ask me if I want to allow the installer to write to the protected file system, clicked yes, and everything worked without problems.

Interesting. Did you get the message stating something to the extent of "This program is not compatible with this version of MacOS." and have to press "Install Anyway"?

> I wonder if this has to do with relocation execution (running installer from where it was downloaded vs downloading it, moving it, then running it).
>
> macOS will do additional relocation sandboxing [...]

It's an Admin account and the only one on the system. Interesting about macOS's sandboxing. I was running it from the Downloads folder, but just tried moving it to the desktop and got the same result. I deleted it and downloaded it again and nothing has changed.

vectorsigma72, 2019-11-06:
Guys, you may be intrested in try  [Clover.app](https://www.insanelymac.com/forum/topic/341047-cloverapp-testing/l)

joshhorsley, 2019-11-08:
The installer fails for as me on 10.15.1 as an Admin immediately after requesting my password to "...make the root file system read-write".

 It gives the error message: "Cannot install without ensuring the file system is read-write". Doesn't seem to matter if I've run the file from the Downloads folder or relocated it.

augiem, 2019-11-08:
> The installer fails for as me on 10.15.1 as an Admin immediately after requesting my password to "...make the root file system read-write".
>
> It gives the error message: "Cannot install without [...]

https://github.com/CloverHackyColor/CloverBootloader/issues/22

closed 2019-11-21 by SergeySlice

augiem, 2019-11-21:
Why was this thread closed? No solution was ever found.