CloverHackyColor/CloverBootloader #27: v2.5 kr5098 - Does not create EFI partition with Catalina, resulting media not bootable

alshdavid, 2019-11-04:
With the following settings, the resulting USB has no EFI partition and is not bootable.
The EFI folder appears on the main data partition.

I have a Macbook Pro 13 2019 running Mojave
[image]

[image]

t14462, 2019-11-04:
Is it real MacBook, or just SMBios? You don't need Clover for real Apple devices, because it's unsupported and can damage your mac

alshdavid, 2019-11-04:
I'm running the clover installer on a MacBook but targeting a USB drive to make it bootable.

So I create the installer on a USB with:
```
sudo /Applications/Install\ macOS\ Mojave.app/Contents/Resources/createinstallmedia --volume /Volumes/MYUSBDRIVE
```

Then ran the aforementioned configuration on the USB

alshdavid, 2019-11-04:
Ah, I had to format the USB with a GUID partition map

closed 2019-11-04 by alshdavid