CloverHackyColor/CloverBootloader #35: Scan entries issue on r5070

Riko-Shet, 2019-11-15:
I have a freez on Scan entries. I figured out it's one of the 4 hdd that I have in raid0 in OS using SoftRaid software. It was working fine before update from r4xxx. What I'm doing: take off the data cable from that hdd, when clover gui is loaded I insert back the data cable and hit enter. So, once I'm on desktop the SoftRaid software shows me SMART status without any error on any of 4 hdds.
What can I provide guys to show you the problem? The problem with logs is that clover is freezed so have no idea what to do.
Thank you!

SergeySlice, 2019-11-15:
If Clover started then debug.log can be produced
config.plist->Boot->Debug=YES

SergeySlice, 2019-11-26:
Closed as abandoned

closed 2019-11-26 by SergeySlice