CloverHackyColor/CloverBootloader #39: Hang on boot v2.5-5098

beefstew, 2019-11-19:
I upgraded Clover from maybe 4722 (am unsure of specific version) to 5098 and now the machine won't boot. Clover menu appears, I select typical boot entry (or let it select based on timeout), bunch of EFI-related boot logs display, and process hangs after a row of +++++++++

[image]

I had overwritten my working rescue USB thumb drive, so I built a new one using Mojave 10.14 installer following the [vanilla desktop guide](https://hackintosh.gitbook.io/-r-hackintosh-vanilla-desktop-guide/building-the-usb-installer). I am using the exact same thumb drive that I used for initial installation a few years ago; drive appears to work just fine. Booting from USB drive also results in errors or hang.

I have tried many configuration variants of drivers & kexts. For example, with and without AptioMemoryFix and/or AptioInputFix. For me, installing AptioMemoryFix cause what seems to be memory-related errors along the lines of:
```
Error allocating 0x1a0 pages at 0x000000000ff62000 alloc typ 2
Couldn't allocate device tree
```

I have also tried older versions of CloverBootloader and either experience same hang after ++++++++++ or errors that look memory-related.

I do not ever see a debug log created in `EFI/CLOVER/misc`.

The computer's installation is also 10.14. Computer hardware is a Gigabyte [GB-BXi3-5010](https://www.gigabyte.com/Mini-PcBarebone/GB-BXi3-5010-rev-10#ov). Even though BIOS settings were working, I updated settings to mirror recommendations [recommendations for Haswell](https://hackintosh.gitbook.io/-r-hackintosh-vanilla-desktop-guide/config.plist-per-hardware/haswell) in vanilla guide.

Any suggestions?

Configuration for latest attempt attached: [CLOVER.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3861662/CLOVER.zip)

SergeySlice, 2019-11-21:
Anyway debug log must be.

beefstew, 2019-11-21:
Earlier, I was not actually botting from the USB device. I have been re-performing testing with alternate configurations now that I'm booting from the correct device.

Debug log shows what looks to be a functioning Clover. Am still experiencing  the hang after End RandomSeed ++++

[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/3875380/debug.log)

I searched Clover source code, trying to determine what exactly this step is doing (since log terminates prior to this). I didn't find an answer (nor any documentation of the boot process), but think that this is the point where the bootloader has loaded the kernel and its extensions into memory hands over control. Because the hang appears to be instantaneous, I'm suspecting a memory address jump failure.

SergeySlice, 2019-11-26:
You may find the line "++++" in Clover sources.
After that Clover does DSDT, kext and kernel patches and jump to kernel.
For my experience there is no reason for wrong memory address to jump.
You may switch on kext patching debug and see the process on the screen then decided if the process hang or kernel hang.

SergeySlice, 2019-11-26:
I see you are using CsmVideoDxe.efi. Try to boot without it. It is poorly working with Intel Graphics.

SergeySlice, 2019-11-26:
This line is not good

> 16:734  0:022  Board-ID set from config as Mac-CENSORED

:)

beefstew, 2019-12-13:
Thanks for your help, SergeySlice.

1) I manually censored that Board-ID line prior to upload -- it's a valid value in the real file
2) I removed the CsmVideoDxe.efi driver
3) Upgraded to 5100

Between 2 and 3, the machine finally boots. Still no sound (original reason for upgrade), but I have a functioning computer again.

[CLOVER-working.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3958923/CLOVER-working.zip)

closed 2019-12-13 by beefstew