CloverHackyColor/CloverBootloader #40: v2.5k-5098 installer error on macOS 10.15.1 (repost)

augiem, 2019-11-21:
I am reposting this issue because it was closed yesterday without any resolution. The original issue is here: https://github.com/CloverHackyColor/CloverBootloader/issues/26

The latest release will not install on macOS 10.15.1, failing with the error "An unexpected error occurred while moving files to the final destination."

Log:
https://pastebin.com/LsSq9wnR

Excerpt:
Nov 2 15:19:20 #####s-iMac shove[1446]: [source=file] failed _RelinkFile(/Library/InstallerSandboxes/.PKInstallSandboxManager/903C5760-622B-45B8-A2F1-5EAB219523B4.activeSandbox/Root/EFIROOTDIR/EFI/BOOT/BOOTX64.efi, /EFI/BOOT/BOOTX64.efi): Directory not empty

I've tried:
Disabling SIP
running sudo mount -uw /

vectorsigma72, 2019-11-22:
Hi can you give a try to [Clover.app](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5099/Clover.app-5099.zip)?? Is in Beta stage but is stable enough. Created to get rid of all problems of the pkg.
More info [here](https://www.insanelymac.com/forum/topic/341047-cloverapp-testing/)

augiem, 2019-12-06:
> Hi can you give a try to [Clover.app](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5099/Clover.app-5099.zip)?? Is in Beta stage but is stable enough. Created to get rid of [...]

Hi,

I was able to install it using the tool. Sorry about not trying it before. I didn't realize you were a contributor to Clover and I just assumed it was some unrelated 3rd party tool and I didn't want to have to go that route just to install it. Thanks for your help.

closed 2019-12-06 by augiem