CloverHackyColor/CloverBootloader #44: Request csmvideodxe / edp / lvds

thefiredragon, 2019-11-24:
Dear,

I have a request, I noticed some changes on my new notebook, so clover can't detect my screen resolution and screen size.

I don't know if this is an issue, I tried to change gui settings and played around with csmvideodxe.efi
Without csmvideodxe external screens over dp not working.

Also I noticed that I run an opensuse installer the can grub detect right screen size and resolution.
Think clover is here for edp connected displays unable to detect the edid.
Running on lenovo yoga c940.

Clover is working only at the bottom of the screen I have a black unused area of 3 centimetres.
macOS run with internal screen only without framebuffer loading, or I need to use external screen and disconnect internal screen from mainboard.

best Regards

SergeySlice, 2019-11-26:
Please provide preboot.log

thefiredragon, 2019-11-26:
How I can do that ?

thefiredragon, 2019-11-28:
`0:100  0:100  MemLog inited, TSC freq: 2592065040
0:100  0:000  CPU was calibrated with RTC
0:100  0:000  
0:100  0:000  Now is 28.11.2019,  7:30:58 (GMT)
0:100  0:000  Starting Clover revision: 5098 (master, commit 58649391) on Phoenix Technologies Ltd. EFI
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED --conf=/Users/sergey/src/Clover/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SelfDevicePath=PciRoot(0x0)\Pci(0x14,0x0)\USB(0x0,0x0)\HD(1,GPT,13090220-B592-460E-A9BE-CE276B663DF3,0x28,0x64000) @3413AA98
0:100  0:000  SelfDirPath = \EFI\BOOT
0:106  0:005  SimpleTextEx Status=Success
0:106  0:000  === [ Get Smbios ] ========================================
0:106  0:000  Type 16 Index = 0
0:106  0:000  Total Memory Slots Count = 4
0:106  0:000  Type 17 Index = 0
0:106  0:000  SmbiosTable.Type17->Speed = 2667MHz
0:106  0:000  SmbiosTable.Type17->Size = 8192MB
0:106  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 ChannelA-DIMM0
0:106  0:000  SmbiosTable.Type17->Vendor = Samsung
0:106  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:106  0:000  SmbiosTable.Type17->PartNumber = M471A1G44AB0-CTD    
0:106  0:000  Type 17 Index = 1
0:106  0:000  Ignoring insane frequency value 0MHz
0:106  0:000  SmbiosTable.Type17->Speed = 0MHz
0:106  0:000  SmbiosTable.Type17->Size = 0MB
0:106  0:000  SmbiosTable.Type17->Bank/Device = BANK 1 ChannelA-DIMM1
0:106  0:000  SmbiosTable.Type17->Vendor = <null string>
0:106  0:000  SmbiosTable.Type17->SerialNumber = <null string>
0:106  0:000  SmbiosTable.Type17->PartNumber = <null string>
0:106  0:000  Type 17 Index = 2
0:106  0:000  SmbiosTable.Type17->Speed = 2667MHz
0:106  0:000  SmbiosTable.Type17->Size = 8192MB
0:106  0:000  SmbiosTable.Type17->Bank/Device = BANK 2 ChannelB-DIMM0
0:106  0:000  SmbiosTable.Type17->Vendor = Samsung
0:106  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:106  0:000  SmbiosTable.Type17->PartNumber = M471A1G44AB0-CTD    
0:106  0:000  Type 17 Index = 3
0:106  0:000  Ignoring insane frequency value 0MHz
0:106  0:000  SmbiosTable.Type17->Speed = 0MHz
0:106  0:000  SmbiosTable.Type17->Size = 0MB
0:106  0:000  SmbiosTable.Type17->Bank/Device = BANK 3 ChannelB-DIMM1
0:106  0:000  SmbiosTable.Type17->Vendor = <null string>
0:106  0:000  SmbiosTable.Type17->SerialNumber = <null string>
0:106  0:000  SmbiosTable.Type17->PartNumber = <null string>
0:106  0:000  Boot status=0
0:106  0:000  Running on: '81TE' with board 'LNVNB161216'
0:106  0:000  === [ GetCPUProperties ] ==================================
0:106  0:000  CPU Vendor = 756E6547 Model=906EA
0:106  0:000   The CPU supported SSE4.1
0:106  0:000  BrandString = Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz
0:106  0:000   The CPU supported turbo
0:106  0:000  MSR 0x35               6000C
0:106  0:000   TSC/CCC Information Leaf:
0:106  0:000    numerator     : 216
0:106  0:000    denominator   : 2
0:106  0:000   Calibrated ARTFrequency: 24000602
0:106  0:000   Rounded ARTFrequency: 24000000
0:106  0:000  MSR 0xE2 before patch 1E008008
0:106  0:000  MSR 0xE2 is locked, PM patches will be turned on
0:106  0:000  MSR 0xCE              0008083B_F1011A00
0:106  0:000  corrected FLEX_RATIO = 0
0:106  0:000  MSR 0x1B0             00000000
0:106  0:000  FSBFrequency = 96 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:106  0:000  MaxDiv/MinDiv: 26.0/8
0:106  0:000  Turbo: 42/43/44/45
0:106  0:000  Features: 0xBFEBFBFF
0:106  0:000  Threads: 12
0:106  0:000  Cores: 6
0:106  0:000  FSB: 100 MHz
0:106  0:000  CPU: 2475 MHz
0:106  0:000  TSC: 2475 MHz
0:106  0:000  PIS: 100 MHz
[... 755 lines cut ...]
61:337  0:005      Force PlugIn kext: System\Library\Extensions\IONetworkingFamily.kext\Contents\PlugIns\AppleVmxnet3Ethernet.kext
61:342  0:004      Force PlugIn kext: System\Library\Extensions\IONetworkingFamily.kext\Contents\PlugIns\mDNSOffloadUserClient.kext
61:345  0:003  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Other
61:345  0:000  Extra kext: EFI\CLOVER\kexts\Other\XHCI-unsupported.kext (v.0.9.2)
61:348  0:002  Extra kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext (v.2.0.4)
61:361  0:013    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Mouse.kext.dSYM (v.2.0.4)
61:367  0:005    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Mouse.kext (v.2.0.4)
61:372  0:005    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Trackpad.kext.dSYM (v.2.0.4)
61:378  0:005    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Keyboard.kext (v.2.0.4)
61:394  0:016    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Trackpad.kext (v.2.0.4)
61:415  0:020    |-- PlugIn kext: EFI\CLOVER\kexts\Other\VoodooPS2Controller.kext\Contents\PlugIns\VoodooPS2Keyboard.kext.dSYM (v.2.0.4)
61:416  0:000  Extra kext: EFI\CLOVER\kexts\Other\VirtualSMC.kext (v.1.0.9)
61:430  0:014  Extra kext: EFI\CLOVER\kexts\Other\USBInjectAll.kext (v.0.7.1)
61:697  0:266  Extra kext: EFI\CLOVER\kexts\Other\SMCProcessor.kext (v.1.0.9)
61:702  0:004  Extra kext: EFI\CLOVER\kexts\Other\SMCBatteryManager.kext (v.1)
61:707  0:005  Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.3.4)
61:741  0:033  Extra kext: EFI\CLOVER\kexts\Other\SMCSuperIO.kext (v.1.0.9)
61:746  0:004  Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.3.9)
61:776  0:030  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Off
61:776  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15
61:779  0:003  Custom boot screen not used because entry has unset use graphics
61:780  0:000  Closing log
`

here's my preboot.log

SergeySlice, 2019-11-28:
Is it your real screen resolution?
8:702 0:000 SetScreenResolution: 3840x2160 - already set

thefiredragon, 2019-11-28:
Yes it is, but I think I'm going to sell this notebook, At the log I don't
really found any issues.

But I can only boot with a fake ID.
I know the Display is connected over edp and I read really a lot on some
notebooks it's working by disabling the alwaysconnected flag.

I determinated that the Display will be found on ioreg with right Index 0
busid 0x00 and connector 02000000 (lvds) without disabling this flag.
Unfortunately screen stay black and after 2 mins system force reboots.
(Used second monitor)

Also tried to use and edid override with forced RGB.
No luck also here.

It's a nice machine but out of my knowledge and without screen this
notebook makes no sense. But all newer Lenovo devices are connected over
edp so I'm afraid if it could work on other machines.

On my Yoga 730 all works fine but 13" Display is to small for daily work
for me ;)

Thanks for your feedback and good work with clover.
I think it's Not a clover issue. More a missconfiguration or Display is not
supported under mac

Best Regards Dan

closed 2019-12-30 by SergeySlice