CloverHackyColor/CloverBootloader #49: Clover Bootloader no connection

Emanuele-1998, 2019-12-10:
why was the recovery connection blocked with recent clovers? in fact the connection in recovery is had only with the 5092 version?

roddy20, 2019-12-10:
what is Recovery Connection?
reinstall macOS from Recovery works, Online Help works

Emanuele-1998, 2019-12-10:
https://imgur.com/a/n3wXktx

Connection in recovery doesn’t work even if the ethernet driver is present, in the 5092 version it works as in this another photo 

if the ethernet driver is present, in the 5092 version it works

https://imgur.com/a/q0VPi3m

Emanuele-1998, 2019-12-10:
> what is Recovery Connection?
> reinstall macOS from Recovery works, Online Help works

Connection in recovery doesn’t work even if the ethernet driver is present as in this photo https://imgur.com/a/n3wXktx , in the 5092 version it works as in this another photo 

https://imgur.com/a/q0VPi3m

MysticalOS, 2019-12-11:
5093+ had
"More correct kext injection for Recovery and installation mode."

so if you have specific testing that shows 5092 is last working version for your use case, that's a pretty spot on where problem could be. I suspect Slice is going to want as many logs as you can provide from a verbose boot though with logging enabled. As well as WHAT ethernet kext you are trying to inject. intel or Atheros

roddy20, 2019-12-11:
even if the ethernet driver is present 

what driver do you use? Realtek 8169 works, Yukon Marwell works

Emanuele-1998, 2019-12-11:
> 5093+ had
> "More correct kext injection for Recovery and installation mode."
>
> so if you have specific testing that shows 5092 is last working version for your use case, that's a pretty spot on where [...]

yes I assure you that it works because to make the usb from windows I use the recovery to download macos without using the virtual machine and I was told by the developer of the script to use the 5092. in fact after the 5092 the ethernet no longer works or wifi as an injection in the recovery mode.

https://github.com/corpnewt/gibMacOS/issues/42#issuecomment-554765695

how do I specify tests to show that from the version subsequent to 5092 the inextention of kext ethernet and wifi no longer works?

Emanuele-1998, 2019-12-11:
> even if the ethernet driver is present
> 
> what driver do you use? Realtek 8169 works, Yukon Marwel works, WiFi Atheros works
 
I use intelmausiethernet

Emanuele-1998, 2019-12-11:
> 5093+ had
> "More correct kext injection for Recovery and installation mode."
>
> so if you have specific testing that shows 5092 is last working version for your use case, that's a pretty spot on where [...]

I use the same computer for testing. 

Verbose in 5092

https://tlgur.com/d/GX3WddB8

Verbose in 5100

https://tlgur.com/d/GPZWXr04

Emanuele-1998, 2019-12-13:
need anything else? @MysticalOS @roddy20

roddy20, 2019-12-13:
> need anything else? @MysticalOS @roddy20

nothing special
installed Clover 5100 and it work as before

Emanuele-1998, 2019-12-13:
> nothing special
> installed Clover 5100 and it work as before

No but in the video in clover 5100 doesn’t work the connection while In clover 5092 work connection in recovery mode

CMMChris, 2019-12-16:
I can confirm there is some kind of issue regarding Ethernet connections in the Recovery. It only seems to affect the RTL8111 driver though because machines needing this driver are the only ones I personally have seen the problem occuring on.

I did a couple of Hackintosh tutorials for some Chinese mini PCs on Youtube (look for Beelink U55, Beelink L55, Chuwi GTBox and Vorke V8). As of recently, a lot of people (got 12 complaints so far) using the Windows based installation method with cVad's Boot Disk Utility which creates a recovery stick for the macOS installation report they can't connect to the internet. However, after macOS is installed, Ethernet is fully working inside the system. So only recovery is affected.

The issue is really strange tho because I haven't been able to nail the cause of the issue so far. Neither different versions of macOS, Clover nor the ethernet kext seem to bring it back to life.

On my workhorse hackintosh with Intel based ethernet it works as usual in recovery.

Emanuele-1998, 2019-12-17:
> I can confirm there is some kind of issue regarding Ethernet connections in the Recovery. It only seems to affect the RTL8111 driver though because machines needing this driver are the only ones I [...]

at least I'm not the only one having this problem in recovery of the nonexistent connection from the version after to the 5092. I hope there will be a patch in the next clover bootloader. @SergeySlice

Emanuele-1998, 2019-12-18:
I hope that @SergeySlice  updates clover bootloader to solve this problem with later versions

CMMChris, 2019-12-23:
Did anyone test the latest Clover release? Don't have a machine with Realtek Ethernet here at the moment.

roddy20, 2019-12-23:
> Did anyone test the latest Clover release? Don't have a machine with Realtek Ethernet here at the moment.

Yes, I have, and yes, it works
PCI\VEN_10EC&DEV_8167&SUBSYS_820D1043&REV_10
I can reinstall macOS from its Recovery Environment by network

CMMChris, 2019-12-23:
Well, you previously claimed you don't have the issue so your network card wasn't affected by it. Thus your result unfortunately doesn't say anything about the issue being fixed or not.

jvpodrouzek, 2019-12-23:
Tried it just now, doesn't work, the interface information box is again empty. RTL8111 and I tried IntelMausi few days ago on my laptop which also didn't work. Version 5092 works on both.

CMMChris, 2019-12-23:
Thanks for clarifying. So we have to keep waiting.

Emanuele-1998, 2019-12-25:
> > Did anyone test the latest Clover release? Don't have a machine with Realtek Ethernet here at the moment.
> 
> Yes, I have, and yes, it works
> PCI\VEN_10EC&DEV_8167&SUBSYS_820D1043&REV_10
> I can reinstall macOS from its Recovery Environment by network

Merry Christmas, what ethernet do you mean the brand? and what kext do you use?

Emanuele-1998, 2019-12-25:
> Tried it just now, doesn't work, the interface information box is again empty (RTL8111). I tried IntelMausi few days ago on my laptop which also didn't work. Version 5092 works on both.

Yes I use always 5092 for the internet in recovery but @SergeySlice I hope he will answer and fix it

roddy20, 2019-12-25:
> VEN_10EC&DEV_8167

VEN_10EC&DEV_8167 is Realtek https://pci-ids.ucw.cz/read/PC/10ec/8167
driver com.apple.driver.AppleRTL8169Ethernet (1.1.4a5)

CMMChris, 2019-12-30:
New release. Any improvements for RTL8111 in Recovery?

Emanuele-1998, 2019-12-30:
> New release. Any improvements for RTL8111 in Recovery?

Yes finally

closed 2019-12-30 by SergeySlice

CMMChris, 2020-01-11:
Issue is not fixed for all cases it seems. I can confirm this issue still appears on some machines running on the lastest Clover releases and others report the same. For example here: https://www.hackintosh-forum.de/forum/thread/46049-recovery-bootstick-kein-internet/

Urgeistenergie, 2020-01-11:
Unfortunately the problem remains. Doesn't work in latest Clover 5103, with "Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 03)" with RealtekRTL8111.kext (v2.2.2).
But works fine with: Clover 5092.

deoeztuerk, 2020-02-03:
the problem still remain with the newest clover.
i took me 2 days to realize the problem.
I unscrewed all my devices and changed the networkcard.

It also affects the dw1820a.