CloverHackyColor/CloverBootloader #50: Clover.app installation fail

CapnDoody, 2019-12-10:
Honest not sure if i should be selecting the EFI partition or the "MacOS Installer" partition, but either one results in this error message:
Installation begin..
0:230: 
execution error: My Path = /private/var/folders/6x/cgj9ntt53bjfkwtxct41m3800000gn/T/AppTranslocation/87D3CDD4-B8CC-495A-BB12-88A2DB533AFB/d/Clover.app/Contents/MacOS/Cloverhelper
Target volume: /Volumes/EFI
Error: target volume "/Volumes/EFI" is not writable. (1)

I have tried with the EFI partition mounted and unmounted

CapnDoody, 2019-12-10:
https://imgur.com/zhsSEEb
https://imgur.com/tbGYegJ

orospakr, 2019-12-12:
A test of target volume writability appears to be failing at https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverApp/cloverhelper/Cloverhelper/main.m#L169.

It's using NSFileManager to check if the EFI volume is writable.  https://developer.apple.com/documentation/foundation/nsfilemanager/1416680-iswritablefileatpath?language=objc

I did confirm that my target EFI volume is mounted read-write.

Presumably this has to do with security enforcement on Catalina.  I don't think I was given the "use removable volume" permission prompt.

A word about my use case: I'm using Clover.app to install Clover on a USB for use on an entirely different machine; I have no interest in disabling SIP or anything else on machine I'm trying to use to set up the USB.

vectorsigma72, 2019-12-13:
> I have tried with the EFI partition mounted and unmounted

Hi, I need you to run the command line attached to debug the problem:

`sudo writability /Volumes/EFI`
where '/Volumes/EFI' must be the mount point of your ESP on your 'sandisk ultra usb 3.0 media' or what ever the mount point is.
Let me know the output.
[writability.zip](https://github.com/CloverHackyColor/CloverBootloader/files/3962013/writability.zip)

vectorsigma72, 2019-12-13:
> execution error: My Path = /private/var/folders/6x/cgj9ntt53bjfkwtxct41m3800000gn/T/AppTranslocation/87D3CDD4-B8CC-495A-BB12-88A2DB533AFB/d/Clover.app/Contents/MacOS/Cloverhelper

The above path doesn't looks good and I think the problem is that. Why the app is at '/private/var/folders/6x/cgj9ntt53bjfkwtxct41m3800000gn/T/AppTranslocation/87D3CDD4-B8CC-495A-BB12-88A2DB533AFB/d/' ???????

orospakr, 2019-12-13:
@vectorsigma72 aha! that's a good clue.

I just went a-googling and found this: https://lapcatsoftware.com/articles/app-translocation.html

I think I was running it from my downloads directory, too.

vectorsigma72, 2019-12-13:
> @vectorsigma72 aha! that's a good clue.
> 
> I just went a-googling and found this: https://lapcatsoftware.com/articles/app-translocation.html
> 
> I think I was running it from my downloads directory. I bet @CapnDoody was, too.

Looks a Gatekeeper problem to me.the app is not code signed, but I think the OS already warn you about that. Run in Terminal:
`sudo spctl --master-disable`
should solve the issue. But can you run the writability command line to ensure the code is good?

orospakr, 2019-12-13:
Alas I'm not at that machine right now and won't be until at least Sunday. But I'll have a look, then!

vectorsigma72, 2019-12-13:
> Alas I'm not at that machine right now and won't be until at least Sunday. But I'll have a look, then!

In the mean time I'll add an alert window to let the user knows if the app is the quarantine directory and so let the user  move the app some where else.

tgoeke, 2019-12-25:
I have a seemingly similar issue.  Using r5101.  Trying to build an installer USB from my MacBook Pro.
Running the pkg complains about the SIP issue.  Running the Clover.app caused Mac security to stop the app, even after allowing the app in security prefs, it does not launch and run.

I'm on Catalina 10.15.2 running a MacBook Pro (15-inch, 2018) 2.9 GHz 6-Core Intel Core i9.

I'm trying on another Mac shortly, I'll use something that is more suitable to disabling SIP.

vectorsigma72, 2019-12-27:
> I have a seemingly similar issue. Using r5101. Trying to build an installer USB from my MacBook Pro.
> Running the pkg complains about the SIP issue. Running the Clover.app caused Mac security to stop [...]

Right click with the mouse and click open because you have to authorize it. Then move the app out of the Downloads directory, please.

JacoGasp, 2019-12-31:
> Honest not sure if i should be selecting the EFI partition or the "MacOS Installer" partition, but either one results in this error message:
> Installation begin..
> 0:230:
> execution error: My Path = [...]

Same problem for me. Cannot install on a USB driver with error:
```
Installation begin..
0:131: 
execution error: My Path = /Applications/Clover.app/Contents/MacOS/Cloverhelper
Target volume: /Volumes/EFI 1
Error: target volume "/Volumes/EFI 1" is not writable. (1)
```

vectorsigma72, 2020-01-01:
> Same problem for me. Cannot install on a USB driver with error:
>
> ```
> Installation begin..
> 0:131:
> execution error: My Path = /Applications/Clover.app/Contents/MacOS/Cloverhelper
> Target volume: [...]

Please try v1.13 where I removed the check.
[Clover.app-v1.13Beta-5102.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4014285/Clover.app-v1.13Beta-5102.zip)

tothandras, 2020-01-02:
@vectorsigma72 I'm getting the following error with the beta:

```
Checking files...
Installation begin..

0:131: execution error: Clover Installer r5102 (1d7e7327), 2020-Jan-02 16:59:11

My Path = /Applications/Clover.app/Contents/MacOS/Cloverhelper
Target volume: /Volumes/EFI
Error Domain=NSCocoaErrorDomain Code=513 "You don’t have permission to save the file “CLOVER” in the folder “EFI”." UserInfo={NSFilePath=/Volumes/EFI/EFI/CLOVER, NSUnderlyingError=0x7fac09704d00 {Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted"}} (1)

Installation failed.
```

I was able to create a folder manually on the volume (in Finder).

vectorsigma72, 2020-01-02:
> @vectorsigma72 I'm getting the following error with the beta:
> 
> ```
> Checking files...
> Installation begin..
> 
> 0:131: execution error: Clover Installer r5102 (1d7e7327), 2020-Jan-02 16:59:11
> 
> My Path = /Applications/Clover.app/Contents/MacOS/Cloverhelper
> Target volume: /Volumes/EFI
> Error Domain=NSCocoaErrorDomain Code=513 "You don’t have permission to save the file “CLOVER” in the folder “EFI”." UserInfo={NSFilePath=/Volumes/EFI/EFI/CLOVER, NSUnderlyingError=0x7fac09704d00 {Error Domain=NSPOSIXErrorDomain Code=1 "Operation not permitted"}} (1)
> 
> Installation failed.
> ```
> 
> I was able to create a folder manually on the volume (in Finder).
> 
> I've tried to give the app full disk access, but I got the same error.

You were able to create a directory after the app mounted the ESP or did you manage to mount the ESP by your self too? 
"Operation not permitted" means no root permissions I guess. Are you running from an Administrative account or else? 
Please do this test:
mount the ESP before open the installer and then try to install on it.

jwmann, 2020-01-04:
I'm also having the same exact issue on 10.15.1
I first tried with v1.12r5102 without success. Same "is not writable." error
I then tried it with the proposed v1.13 zip and received the same "Operation not permitted" error as well.

I tried mounting the EFI volume prior to running the install and it resulted in the same issue.

vectorsigma72, 2020-01-04:
I've swiched to a different technology in v1.14 i.e. from NSFileManager from the obj-c runtime to FileManager class from swift 5. Any one with this problem care to try? Thanks in advance.

[Clover.app-v1.14Beta-5102.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4022133/Clover.app-v1.14Beta-5102.zip)

tothandras, 2020-01-04:
Thanks! I've tried it. Similar error:

```
Checking files...
Installation begin..

0:143: 
execution error: Clover Installer r5102 (aa207eaa) (v1.1.0), 2020-01-04 06:32:00
SELF = /Applications/Clover.app/Contents/MacOS/CloverDaemonNew
Target volume: /Volumes/EFI
Error: target volume "/Volumes/EFI" is not writable. (1)

Installation failed.
```

I'm using an administrator account and the EFI volume is mounted.

vectorsigma72, 2020-01-04:
Ok thanks. Can you post the result of:
`diskutil info /Volumes/EFI`
just in case I can found additional infos..

tothandras, 2020-01-04:
```
➜ diskutil info /Volumes/EFI
   Device Identifier:         disk2s1
   Device Node:               /dev/disk2s1
   Whole:                     No
   Part of Whole:             disk2

   Volume Name:               EFI
   Mounted:                   Yes
   Mount Point:               /Volumes/EFI

   Partition Type:            EFI
   File System Personality:   MS-DOS FAT32
   Type (Bundle):             msdos
   Name (User Visible):       MS-DOS (FAT32)

   OS Can Be Installed:       No
   Media Type:                Generic
   Protocol:                  USB
   SMART Status:              Not Supported
   Volume UUID:               0E239BC6-F960-3107-89CF-1C97F78BB46B
   Disk / Partition UUID:     4D94E4CC-B5B4-440A-BF0B-2E5A268F052D
   Partition Offset:          20480 Bytes (40 512-Byte-Device-Blocks)

   Disk Size:                 209.7 MB (209715200 Bytes) (exactly 409600 512-Byte-Units)
   Device Block Size:         512 Bytes

   Volume Total Space:        206.5 MB (206472192 Bytes) (exactly 403266 512-Byte-Units)
   Volume Used Space:         814.1 KB (814080 Bytes) (exactly 1590 512-Byte-Units) (0.4%)
   Volume Free Space:         205.7 MB (205658112 Bytes) (exactly 401676 512-Byte-Units) (99.6%)
   Allocation Block Size:     512 Bytes

   Read-Only Media:           No
   Read-Only Volume:          No

   Device Location:           External
   Removable Media:           Removable
   Media Removal:             Software-Activated

   Solid State:               Info not available
```

vectorsigma72, 2020-01-05:
Not sure what it is going on because any disks you can select in the UI was added because marked as writable by the DiskArbitration framework. So somehow this is later marked read only by mount, I guess. The attached version (commit 2e4cbf3) try to make the target volume as read-write (if found read only). Please try and let me know, thanks in advance
[Clover.app-v1.14Beta-5102.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4023463/Clover.app-v1.14Beta-5102.zip)
.

tothandras, 2020-01-05:
```
Checking files...
Installation begin..

0:143: 
execution error: Clover Installer r5102 (f236554b) (v1.1.1), 2020-01-05 07:56:43
macOS Version 10.15.2 (Build 19C57)
SELF = /Applications/Clover.app/Contents/MacOS/CloverDaemonNew
Target volume: /Volumes/EFI
BSD Name:      disk2s1
Filesystem:    FAT32
Is internal:   false
Is writable:   true
o_Ops: '/Volumes/EFI' appear to be not writable, remounting read/write.
sleeping 6 seconds..
Error: target volume "/Volumes/EFI" is not writable. (1)

Installation failed.
```

I'm happy to jump on a call with you to debug this. Please let me know if it would help. Thank you!

vectorsigma72, 2020-01-06:
I'm out of ideas, but should be nice to know if that happens only with this stick or if happens only on external drives and/or works or not on internal ones.

tothandras, 2020-01-06:
Unfortunately, I won't be able to test that. :( I'm trying to make a clover usb stick on my macbook for my hackintosh.

tothandras, 2020-01-06:
Unfortunately, I won't be able to test that. :( I'm trying to make a clover usb stick on my macbook for my hackintosh.
What about you @jwmann?

SergeySlice, 2020-02-11:
User's mistake.

closed 2020-02-11 by SergeySlice