CloverHackyColor/CloverBootloader #54: English translation for this precious guide

1alessandro1, 2019-12-13:
As done by acidanthera for Opencore, they they improved their documentation providing a PDF where they can group all knowledge spread across multiple forums and describe all the changes from commit to another

Is it possible an english version for future releases of this previous guide in order for our better understanding? I'm referring to this document:
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5097/Clover.of.Hacky.Color_5097_rus.pdf

It would also be perfect to have a "Differences.pdf" - if it's possible to maintain this guide up-to-date. I strongly believe that this project will benefit whether if is going to happen 

Thank you

NyaMisty, 2020-02-03:
Google translate?
Seems that google translate is enough to understand this document

1alessandro1, 2020-02-16:
@NyaMisty 
Unfortunately I cannot rely on the errors done by Google Translate to have a precise understanding of the document. 
By now, everything on the internet should be exposed to the public in English, so everyone can benefit and understand what's going on.

Also, there's no clear trace on what features are either implemented or removed, as Differences.pdf does with acidanthera. 

https://github.com/acidanthera/OpenCorePkg/blob/master/Docs/Differences/Differences.pdf

closed 2020-04-29 by asava

asava, 2020-05-01:
We prefer not to have documentation as part of open issues. It was tagged as documentation, and will be taken care of, when we get it to it. Thank you for your concern.