CloverHackyColor/CloverBootloader #56: How does Clover work under the hood?

Mimoja, 2019-12-14:
I am trying to understand the different parts that are playing together here.
- How much does clover rely on the underlying UEFI?
- What is rEFIt's role?
- What other dependencies are there?
- What does clover implement exaclty?

I would like to build kexec into something mac compatible from within linux but am struggeling understanding clovers parts.

SergeySlice, 2019-12-16:
I am not clear on your question but I will try to answer.
1. Clover uses some protocols from outer UEFI: GraphicsOutputProtocol, Console, BlockIo, DevicePath and many others. There is a long list.
2. rEFIt is old application which sources used to create Clover GUI as a basic idea.
3. Clover uses sources of EDK2.
4. Clover implements some algorithms to adjust BIOS settings like DMI, ACPI and device properties for our needs. It is also huge story.

Mimoja, 2019-12-16:
Thanks. This already helps a lot.
Is there a rough diagram of the bootflow? Eg. Which component calls into which next stage?

SergeySlice, 2019-12-23:
Legacy boot:
BIOS->HDD MBR block where boot0af written -> ESP block 1 where boot1f32 writen -> boot file on ESP which is CloverEFI based on DUET project -> CloverX64.efi which is boot manager searching for loaders for OSes (macOS, Windows, Linux) -> the bootloader chosen -> OS

UEFI boot:
UEFI BIOS -> CloverX64.efi -> same flow

closed 2019-12-30 by SergeySlice