CloverHackyColor/CloverBootloader #58: Clover Bootloader file system is not writable

Emanuele-1998, 2019-12-18:
the application cannot be run without making sure the file system is writable

[image]

roddy20, 2019-12-18:
Use Clover.app, not Install***.pkg
Or copy EFI folder

Emanuele-1998, 2019-12-18:
> Use Clover.app, not Install***.pkg
> Or copy EFI folder to its place

why can't i use the .pkg but clover.app?

roddy20, 2019-12-18:
you may use anything, but Clover.app is more suitable for new versions of macOS

Emanuele-1998, 2019-12-18:
> you may use anything, but Clover.app is more suitable for new versions of macOS

and from the clover bootloader releases what should I download to find the Clover.app?

roddy20, 2019-12-18:
> what should I download to find the Clover.app?
the first link on that page
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5100/Clover.app-v1.11-5100.zip

roddy20, 2019-12-18:
> the application cannot be run without making sure the file system is writable
> 
try `sudo mount -uw /`
to make it writable

closed 2019-12-18 by Emanuele-1998