CloverHackyColor/CloverBootloader #64: ./buildme Fails Under Catalina

sdhaene, 2019-12-26:
Checking out the branch 5101 and running a `./buildme ` command yields the following error:   
Error compiling nasm-2.14.02 ! Checking the log:

> /Users/sdhaene/Projects/CloverBootloader/toolchain/tools/logs/nasm.make.log.txt:

A number of linking errors occured:
```
...skipping...
make:       _main in nasm.o
*** [rdoff/rdf2bin] Error 1  "_symtabInsert", referenced from:

  "_src_free", referenced from:
      _processmodule in ldrdf.o
      _main in nasm.o
make:   "_globalbits", referenced from:
      _main in nasm.o
*** [rdoff/rdfdump] Error 1
      _main in ldrdf.o
  "_rdfloadseg", referenced from:
      _processmodule in ldrdf.o
      _main in ldrdf.o
  "_add_seglocation", referenced from:
      _main in ldrdf.o
  "_rdfnewheader", referenced from:
      _main in ldrdf.o
ld: symbol(s) not found for architecture x86_64
  "_nasm_free", referenced from:
      _main in ldrdf.o
  "_done_seglocations", referenced from:
      _main in ldrdf.o
clang: error:   "_rdfgetheaderrec", referenced from:
linker command failed with exit code 1 (use -v to see invocation)      _processmodule in ldrdf.o

      _main in ldrdf.o
  "_nasm_malloc", referenced from:
      _processmodule in ldrdf.o
      _main in ldrdf.o
      _loadmodule in ldrdf.o
  "_nasm_strdup", referenced from:
      _processmodule in ldrdf.o
      _main in ldrdf.o
      _loadmodule in ldrdf.o
  "_symtabFind", referenced from:
      _processmodule in ldrdf.o
      _main in ldrdf.o
  "_rdl_openmodule", referenced from:
      _main in ldrdf.o
  "_fwriteint32_t", referenced from:
      _main in ldrdf.o
ld: symbol(s) not found for architecture x86_64
make: *** [rdoff/rdflib] Error 1
clang: error: linker command failed with exit code 1 (use -v to see invocation)
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [nasm] Error 1
make: *** [rdoff/ldrdf] Error 1
```
Also tried the commands `./buildme XCODE8` and `./buildme XCODE11` yield the same results.
Am I doing something wrong, or does the build fails under Catalina?
Using OSX 10.15.2, XCode 11.3 with the additional tools.

Any help would be greatly appreciated.
[nasm.make.log.txt](https://github.com/CloverHackyColor/CloverBootloader/files/4002423/nasm.make.log.txt)

vectorsigma72, 2019-12-27:
Hi, just tried and nasm perfectly build fine in 10.15.2 and Xcode 11.3. Not sure what can be the problem for you.

sdhaene, 2019-12-29:
I've got it working now.
Had to remove Xcode completely, reinstall it, with the "Xcode Extensions" and then mysteriously everything worked.
Did not see any difference in Xcode version whatsoever.

closed 2019-12-29 by sdhaene