CloverHackyColor/CloverBootloader #68: Remove Clover Bootloader (accidentally installed on Mac instead of USB drive)

joneslloyd, 2019-12-31:
Hi – Thanks for creating Clover Bootloader!

I was just wondering what the best / safest was is to remove Clover Bootloader from my MacBook Pro?

I was creating a boot USB drive and accidentally set the install location as my HD instead of the USB drive.

I haven't rebooted the Mac yet, and was wondering what the best way is to remove it.

Thanks in advance!

roddy20, 2019-12-31:
1 it is not a support forum, and there is no clover's problems described in your message
2 the best way to remove is to remove, exactly as removing any other folder

joneslloyd, 2019-12-31:
Thanks for getting back to me – So, this? https://imgur.com/a/egBltuJ (ending in delete)

CMMChris, 2020-01-01:
Don't worry, your Mac won't boot Clover without it being blessed. If you want to remove it just mount your EFI partition and delete the BOOT and CLOVER folders.

roddy20, 2020-01-01:
> Thanks for getting back to me – So, this? https://imgur.com/a/egBltuJ (ending in delete)

is it possible just delete folder without video illustrations ? once again, it is not related with clover development

joneslloyd, 2020-01-01:
Thanks @CMMChris! I just wanted to be safe as I’m unfamiliar with EFI partitions 🙂

closed 2020-01-01 by joneslloyd