CloverHackyColor/CloverBootloader #70: Same as issue #35

Riko-Shet, 2020-01-03:
Hello Slice! Can you please merge that issue with issue #35 ? Thank you!

I have a problem when clover freeze on Scan Entries. Sorry Sergey, was busy and I don't provided the log and you closed the Issue.
Attaching preboot log.
[preboot.log.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4018913/preboot.log.zip)

Is the preboot log reset on every boot? If so you won't find anything there because to boot into the OS I need to de-attach the hdd that gives the problem to clover, wait Clover UI and then hit enter....
Thanks!

SergeySlice, 2020-02-19:
I see no information in the preboot.log. You have to produce debug.log when the issue occurred.
Anyway using a set of Piker's patches may cause such issue.

Riko-Shet, 2020-03-09:
I'm on 5070 now and it's working now. Have no Idea what's happened! We may close that!
Thank you!

closed 2020-03-09 by Riko-Shet