CloverHackyColor/CloverBootloader #71: Package Clover.app-v1.14-5103.zip is damaged.

papasax, 2020-01-07:
And can't be opened.
[image]

esetnik, 2020-01-07:
I saw the same thing but was able to fix it using

`❯ xattr -cr /Applications/Clover.app`

roddy20, 2020-01-07:
it is normal behavior for unsigned app by unknown deleloper
xattr -cr /Applications/Clover.app
or 
Ctrl+Open to start it

papasax, 2020-01-08:
I tried again today and now it's working. Still don't know what happened. (@roddy20 and esetnik : It was not the unknown developer message but the broken app message).
I'm closing the issue anyway.

closed 2020-01-08 by papasax

esetnik, 2020-01-08:
@papasax I had the broken app message the same as your screenshot in the original bug. I think this should be left open as I’ve experienced it with every version of Clover.app.

papasax, 2020-01-08:
@esetnik As far as I'm concerned, I can not reproduce the issue so I'm closing it. 
It could be a github bug. Feel free to re open it if you want.

NiklasBr, 2020-01-08:
@esetnik Are you *sure* the app is broken? Are you sure you are simple not experiencing [this](https://support.apple.com/en-us/HT202491 "How to open an app that hasn’t been notarized or is from an unidentified developer")?

roddy20, 2020-01-08:
> @esetnik As far as I'm concerned, I can not reproduce the issue so I'm closing it.
> It could be a github bug. Feel free to re open it if you want.

it s not a bug, it is system protection feature

esetnik, 2020-01-08:
@niklasbr it is being detected as a malicious app as per the link you provided. So we should at least update the Readme with instructions for bypassing the issue as the Clover app won’t work on any Catalina system straight from a Github download.

[image]