CloverHackyColor/CloverBootloader #72: Clover only installs efi-backup folder and nothing else

leifadev, 2020-01-09:
I am following the Clover Setup portion of the Vanilla Hackintosh guide. I checked all the options that I needed to in the Customize menu. It only installs a folder named "EFI-Backups". I have tried to format my USB according to the guide and tried again. Here are the instructions below I followed.

I will link the guide [here](https://hackintosh.gitbook.io/-r-hackintosh-vanilla-desktop-guide/building-the-usb-installer)
And the clover setup page [here](https://hackintosh.gitbook.io/-r-hackintosh-vanilla-desktop-guide/clover-setup)

The portion of what options to check is quoted here:
> Install Clover for UEFI booting only
Install Clover to the ESP
Under UEFI Drivers:
AptioMemoryFix (the new hotness that includes NVRAM fixes, as well as better memory management)
Note: You can use OsxAptioFix3Drv located in UEFI Drivers -> Memory fix drivers if using the official Clover installer from SourceForge.  Optionally, you can download the final release of AptioMemoryFix from here 
HFSPlus (or VBoxHfs) - one of these is required for Clover to see and boot HFS+ volumes.  If you see the option to enable it in the installer, make sure it's selected - if you don't see it in the installer, verify that one of them exists in your Clover installs efi drivers folder
ApfsDriverLoader - (Available in Dids' Clover builds - or here) this allows Clover to see and boot from APFS volumes by loading apfs.efi from the ApfsContainer located on block device (if using AptioMemoryFix as well, ensure you have R21 or newer)
> 
If you could tell me what I should do that would be great!

real-cereal-boy, 2020-01-16:
You want to map your USB to a diffrent type

real-cereal-boy, 2020-01-16:
Also uncheck one one the options

> HFSPlus (or VboxHFS)

> ApfsDriverLoader

** These options are two different partions ONLY SELECT 1, Depending on your system type you want to choose one **
[image]

SergeySlice, 2020-01-29:
Just a user mistake. There is no Clover issue.

closed 2020-01-29 by SergeySlice