CloverHackyColor/CloverBootloader #74: Clover Recovery Ethernet Connection (Some Realtek Chipsets / #49)

CMMChris, 2020-01-13:
Closed issue #49 is not fixed. There are still multiple reports of non-working Ethernet in Recovery with the RTL8111 Kext on the latest Clover builds. Using 5092 fixes the issue, so definitely caused by Clover Kext injection. Some recent complaints have been added to closed issue #49 .

Florin9doi, 2020-01-16:
Patch for this issue
```diff --git a/rEFIt_UEFI/Platform/kext_inject.c b/rEFIt_UEFI/Platform/kext_inject.c
index dbb172d8..3d951baf 100644
--- a/rEFIt_UEFI/Platform/kext_inject.c
+++ b/rEFIt_UEFI/Platform/kext_inject.c
@@ -109,7 +109,8 @@ BOOLEAN checkOSBundleRequired(UINT8 loaderType, TagPtr dict)
     } else if (OSTYPE_IS_OSX_INSTALLER(loaderType)) {
         if (AsciiStrnCmp(osbundlerequired, "root", 4) &&
             AsciiStrnCmp(osbundlerequired, "local", 5) &&
-            AsciiStrnCmp(osbundlerequired, "console", 7)) {
+            AsciiStrnCmp(osbundlerequired, "console", 7) &&
+            AsciiStrnCmp(osbundlerequired, "network-root", 12)) {
             inject = FALSE;
         }
     }```

closed 2020-01-22 by SergeySlice

deoeztuerk, 2020-02-03:
still present in the newest clover

Florin9doi, 2020-02-03:
This patch is not included in the newest clover release: 5103. You need to compile it yourself or wait for the next release

commit 27de9a9c (2020-01-22): Merge pull request #75 from Florin9doi/issue_74

Repair network installer; Fixes #74
commit 8d011472 (2020-01-22): Repair network installer; Fixes #74