CloverHackyColor/CloverBootloader #76: r5103 hangs at "scan entries"

richardklingler, 2020-01-26:
Good morning

Trying to do a fresh Catalina install on a AsRock H270-Pro4 mainboard I get the "scan entries" message in Clover which stays forever....this is also true for r5102.....on an older release r4961 it scans fine...

Neither using VfsBox.efi nor HfsPlus.efi help...

Attached debug.log....which looks fine to me...

richardklingler, 2020-01-26:
Hmm...see nothing to attach the logs...so put it here:

https://pastebin.com/1iR2JvPS

richardklingler, 2020-01-26:
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4113295/debug.log)

SergeySlice, 2020-01-29:
The hang is after 
13:364  0:391  found AppleSMC protocol
which is controlled by VirtualSMC.efi. 
It is not Clover issue.

closed 2020-01-29 by SergeySlice