CloverHackyColor/CloverBootloader #77: Clover does not select correct volume on 2nd and 3rd reboot during macOS updates

najeebanwer, 2020-01-31:
I have Windows 10, Ubuntu 19.10 and macOS Catalina (10.15.3) installed on a Dell XPS 8910.
The specs are:
CPU: Intel Core i7-6700, Intel HD 530 Graphics
GPU: NVIDIA GeForce GTX 750 Ti 2GB (not being used - not compatible with macOS Catalina)
RAM: 16GB
Clover Version: r5103

In config.plist I have the Boot/DefaultVolume set to LastBootedVolume.

During macOS updates, the system reboots three times. On the first reboot, Clover automatically selects the "macOS Install" volume, which is correct.

However, on the second reboot, Clover appears to "forget" the last booted volume and selects Windows as the volume to boot. I have to manually select the "macOS Install" volume to continue the update process.

After the macOS update completes, Clover selects the macOS Recovery volume to boot. I have to manually select the correct boot volume "Macintosh HD".

I have attached the following files as ZIP archives:
a) config.plist
b) debug.log
c) diskinfo.txt

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4140919/config.plist.zip)
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4140923/Archive.zip)

najeebanwer, 2020-02-01:
I removed my config.plist from the previous post and have attached an edited version with some of the SMBIOS information redacted.
[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4142182/config.plist.zip)

CMMChris, 2020-02-21:
Similar issue as here: https://github.com/CloverHackyColor/CloverBootloader/issues/67
Hopefully this will be addressed soon.

BbIKTOP, 2020-05-13:
The same here

SergeySlice, 2020-05-18:
Is the issue true for 5117?

najeebanwer, 2020-05-18:
Since this issue only happens during a macOS update, I cannot say if this is true for Clover r5117. I will let you know if it happens again during the next macOS update.

najeebanwer, 2020-05-27:
I am using Clover r5118 and updated to macOS 10.15.5.

The issue is partially resolved; Clover now selects the correct volume "macOS Install" during the first and second reboot so the update completes without intervention. 
However, on the third reboot, after the update, it selects the macOS Recovery volume to boot. 

I have attached my debug.log file.
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4686599/debug.log)

I guess now the problem is identical to the one described in issue #67 so this issue can be closed.

BbIKTOP, 2020-05-27:
5118 updated to 10.15.5 ok, boot volume selected correctly on each reboot

closed 2020-05-27 by SergeySlice