CloverHackyColor/CloverBootloader #78: Wrong FSB Speed After Bootup

scorpius, 2020-01-31:
At the boot prompt the FSB is 133Mhz, but once the OS loads the FSB is 533Mhz. This results in the OS running at half speed. I have an i7-860 cpu.

SergeySlice, 2020-02-07:
Show log, please.

scorpius, 2020-02-08:
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/4175196/preboot.log)
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4175198/debug.log)
[image]
[image]

Also enoch boots and the OS runs at normal speed, but various other things do not work, so I prefer clover. I must be missing some option or setting.

SergeySlice, 2020-02-11:
System shows Bus Speed x4 compared to real.
Other issues should be reported separately.

closed 2020-02-11 by SergeySlice