CloverHackyColor/CloverBootloader #79: APFSdriverLoader.efi kill clover GUI

kovacsek, 2020-02-04:
Hi,
I'm trying to figure out why crash Clover GUI on my old PC, when I loaded ApfsDriverLoader.efi driver. This is an P8P67 WS Revolution with an i7-3770 in UEFI mode, with a RX580 Graphics card and a NVMExpress card. I thinks it will be a clover/uefi/memory allocate bug, because this clover EFI folder started with successfully on my thinkpad x230. 

If the driver loaded, freeze after this lines:
```
1:944  0:037  GUI ready
2:726  0:781  Start Tool: \EFI\CLOVER\tools\Shell64U.efi
2:745  0:019  Loading Shell64U.efi  status=Success
```

I thought my EFI folder was corrupted or something, so I created an usb drive, with default setting, but the issue is still standing. Is there any way to obtain more detailed log?

I attach the current log.
[with_apfs.log](https://github.com/CloverHackyColor/CloverBootloader/files/4152638/with_apfs.log)

Thanks.

SergeySlice, 2020-02-07:
For 67 chipset you have to install EnglishDxe.efi. It installs absent protocol needed for Shell.efi.

kovacsek, 2020-02-07:
I tried, but same... is there any way to get or complie for myself a debug version of Shell64U.efi? 
[with_apfs-2.log](https://github.com/CloverHackyColor/CloverBootloader/files/4172719/with_apfs-2.log)

kovacsek, 2020-02-09:
I found a way to solve this problem. I think the bug is in clover.efi but I haven't found yet exactly where... 100% that, this is a firmware(ASUS)&Clover bug because I have another P67 motherboard (MSI version) and it works properly there.

The solution:
I copied the ApfsDriverLoader.efi to the EFI root, and I'm started Clover without this driver. After the GUI was ready, I selected the EFI Shell target, and loaded driver manually:
```
load -nc fsX:\ApfsDriverLoader.efi
connect -r
exit
```

After back to the Clover GUI, there is every APFS drive in the boot list, and bootable.

SergeySlice, 2020-02-11:
Shell64U.efi is already in the Clover release. But you have Clover 4411 which is too old to speak anything.

kovacsek, 2020-02-11:
Hmm... It's maybe user error but now, I don't understand, what's going on. If you can see, there is in 461. line. 

```
460 0:100  0:000  Now is 07.02.2020,  19:37:37 (GMT)
461 0:100  0:000  Starting Clover revision: 5103 (master, commit 3f3b85da0) on American 
```Megatrends EFI

And the next reboot:
```
627 0:100  0:000  Now is 7.2.2020,  19:38:52 (GMT)
628 0:100  0:000  Starting Clover revision: 4411 on American Megatrends EFI
```

I will completly clear my EFI partition, and upload the latest version, and I will create a clear log with these files.

roddy20, 2020-02-11:
> Hmm... It's maybe user error but now, I don't understand, what's going on. If you can see, there is in 461. line.
> And the next reboot:

You have 2 copies of Clover installed on 2 HDDs or HDD and flash drive

kovacsek, 2020-02-12:
So, I clear every EFI partitions and installed the latest version from github. The issue is still standing, but now the log end after "APFS driver loader" in the Load Drivers section.

[5103.log](https://github.com/CloverHackyColor/CloverBootloader/files/4192803/5103.log)

SergeySlice, 2020-02-19:
It is not Clolver issue, it is user's mistake.

SergeySlice, 2020-02-19:
Be sure using latest version of ApfsDriverLoader.
But I think the issue is your system Sierra which contains very old apfs.efi in its container.
You should use apfs.efi instead of ApfsDriverLoader.efi
or reformat apfs container in modern system like Mojave.

kovacsek, 2020-02-25:
I close this ticket. My motherboard dead at the last week, so we probably never know what was the problem.

closed 2020-02-25 by kovacsek

oldman20, 2022-08-23:
I have same problem on my dell3558, I pretty sure my config.plist correctly, almost working!
First time I load manual driver similar above, but..
Finally i can control this issue, can sure reason cause my "special" behavior while load efi drivers. 

All what i did is set ApfsDriverLoader.efi (and HfsPlus.efi, if need for installation macOS with Clover, yes)

Firstly put ApfsDriverLoader.efi and HfsPlus.efi in root of EFI partition. 

Boot into Clover GUI, press S to open EFI Shell:

fsx: #x depend your partitions layout, mine is 0
ls #check for sure x is correct part, if that you will see ApfsDriverLoader.efi and HfsPlus.efi
bcfg driver add y fsx:\ApfsDriverLoader.efi "What ever name"
In my case, I add both ApfsDriverLoader.efi and HfsPlus.efi so these my cmd:

bcfg driver add 0 fs0:\ApfsDriverLoader.efi "Apfs Driver"

bcfg driver add 1 fs0:\HfsPlus.efi "HFS+ Driver" 

Then type: exit to out of EFI She'll/ or just Ctrl+Alt+Del to reboot device, Clover will autoload EFI and show macOS to boot

*NOTE: bcfg will store EFI drivers in NVRAM, so if reset NVRAM will be gone, must repeat above process