CloverHackyColor/CloverBootloader #80: PCIe card efficiency

macattackjack, 2020-02-07:
Two MBs, same exact build with current clover, no ssdt's vanilla build

Asus Rampage IV Gene (x79) i7 3820 (4 core) 16 gb ram ddr3 Mac OS 10.13.6
Pro Tools  application CPU meter, affected directly by its audio card, shows 1% idle load, 4% playing.

Gigabyte Z390 M Gaming board (z390) i9 9900k (8 core) 16 gb ddr4 Mac OS 10.13.6 (The exact same build and SSD)
Pro Tools application CPU meter shows 5% idle load 15% load when playing.

Is this a clover thing, something to do with the efi files I'm loading?

Or different MB makers have different ways of implementing the PCIe busses and some are more efficient than others?

macattackjack, 2020-02-08:
Further info

I did an install of Windows10, and installed the Pro Tools drivers and software

Ran the same test with the same file with both motherboards in windows10, and there was no difference in the cpu meter performance between either motherboard.

It's something on the Mac OS end, and potentially in the efi and kexts?

macattackjack, 2020-02-09:
Probably Solved. Will see when this is tested on an Asus 1151 MB.

3820 has 4 cores, 10 mb L3 Cache, more than 2 per core, 40 PCIe Lanes
9900K has 8 cores, 16 mb L3 Cache, exactly 2 per core, 16 PCIe Lanes

The 9900k just doesn't communicate as fast with the Motherboard via the L3 cache, nor along the PCIe lanes

SergeySlice, 2020-02-11:
Clover has nothing to do with this issue.

closed 2020-02-11 by SergeySlice