CloverHackyColor/CloverBootloader #81: Ivy Bridge (3437U) wrong multiplier detection

Te-eMster, 2020-02-08:
On the 3437U-CPU (in an HP Elitebook Folio 9470m) the bootloader detects the standard multiplier to be 24 (2.4GHz), but it should be 19 (1.9GHz). BIOS detects speed correctly, Clover also shows the correct CPU name, just the multiplier is wrong.

SergeySlice, 2020-02-11:
Show please Clover preboot.log

Te-eMster, 2020-02-11:
[preboot.log](https://github.com/CloverHackyColor/CloverBootloader/files/4184880/preboot.log)

It is with version 4988, but 5103 produces the same result.
Thanks!

SergeySlice, 2020-02-13:
All is good.
1.9GHz is a base frequency.
2.4GHz is max non-turbo frequency. The BIOS started at this frequency.
2.9GHz is max turbo frequency.
If you want to show 1.9GHz in "About This Mac" then you can set 
`	<key>CPU</key>
	<dict>
		<key>FrequencyMHz</key>
		<integer>1900</integer>
`
and enjoy.

closed 2020-02-19 by SergeySlice

Te-eMster, 2020-02-21:
Where did you find that information? I searched all Intel documentation but could not find any mention of 2.4 GHz - just the 1.9GHz non-turbo and the 2.9/2.7 (single/multi-core) turbo.