CloverHackyColor/CloverBootloader #82: Clover.app does not open

steelbrain, 2020-02-17:
Forgive my ignorance but I couldn't find any relevant docs that specified special treatment for Clover.app so I downloaded `Clover.app-v1.15-5104.zip` and tried running on macOS Catalina and this is what I get

[image]

Any ideas what I could be doing wrong? Thanks

vectorsigma72, 2020-02-17:
> Forgive my ignorance but I couldn't find any relevant docs that specified special treatment for Clover.app so I downloaded `Clover.app-v1.15-5104.zip` and tried running on macOS Catalina and this is what I get
> 
> [image]
> 
> Any ideas what I could be doing wrong? Thanks

Hi, try now, it should be ok. I'm not sure why this happen to Slice and not to me. I'll hard code the path to the zip command line shipped with macOS, in case He have some binaries from macports, brew or alike. Also  I'll try to see if something is wrong with the build script as well. We will see.

P.S. Clover binaries included in the new archive are anyway the ones built by Slice.

SergeySlice, 2020-02-18:
I compiled with script buildme->build all in 10.14.6 with xcode 11.3. What is wrong?

SergeySlice, 2020-02-18:
Catalina? I tested in Mojave and all is fine here.

steelbrain, 2020-02-18:
I don't have a Mojave machine to test but it's broken for Catalina. Just tried it again and `Clover.app-v1.15-5104.zip` is still broken just like before

SergeySlice, 2020-02-18:
@vectorsigma72 What external libraries assumed for build Clover.app except macOS10.15SDK?

vectorsigma72, 2020-02-18:
> I don't have a Mojave machine to test but it's broken for Catalina. Just tried it again and `Clover.app-v1.15-5104.zip` is still broken just like before

The cause are attributes written by Xcode at this point. Can you confirm [this](https://www.insanelymac.com/forum/topic/341047-cloverapp-testing/?do=findComment&comment=2708671) fix the issue? .. I can do that just before creating the archive..

vectorsigma72, 2020-02-18:
> @vectorsigma72 What external libraries assumed for build Clover.app except macOS10.15SDK?

It expect swift 5, the one with the stable ABI, introduced in 10.14.4 and Xcode 10.2. Any older will be a no go. Not related, I guess to this issue. I did not expected Apple adds additional attributes in applications not signed w/o a developer id as this is already clear. Just I hope this is not done after you download the archive and decompressed it, which is done by Safari. Easily ensurable by using e.g Chrome, preferably not using the ~Downloads path which I suspect trigs a weird mechanism.

SergeySlice, 2020-02-18:
Made it just now
~~~~
[BUILD APP]
(Require Xcode 11 or greater)
[CHECK XCODE]
Building Clover.app...
[XCODE]

------------------------------------------------------------------------
buildme, Clover v2.5k r5104 (SHA: 20237d492)
TOOLCHAIN: GCC53 (override example: './buildme XCODE8')

~~~
Applied 
xattr -rc Clover.app
Zipped by system archiver. Test, please!
[Clover.app.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4221551/Clover.app.zip)

vectorsigma72, 2020-02-18:
Can you upload a releases to be tested? I'm curious if this can be the fix :-)

vectorsigma72, 2020-02-18:
> Made it just now
>
> ```
> [BUILD APP]
> (Require Xcode 11 or greater)
> [CHECK XCODE]
> Building Clover.app...
> [XCODE]
>
> ------------------------------------------------------------------------
> buildme, Clover [...]

It says it is damaged again :-(

vectorsigma72, 2020-02-18:
`sudo spctl --master-disable`
solve the issue w/o run xattr.

roddy20, 2020-02-18:
> `sudo spctl --master-disable`
> solve the issue w/o run xattr.

it will turn off system security . xattr applies to one program only.
both ways are OK, it is user's preference

vectorsigma72, 2020-02-18:
> > `sudo spctl --master-disable`
> > solve the issue w/o run xattr.
> 
> it will turn off system security . xattr applies to one program only.
> both ways are OK, it is user's preference

You're right, and if fact I wanted to ensure the problem is not a damage archive for true.

roddy20, 2020-02-18:
> You're right, and if fact I wanted to ensure the problem is not a damage archive for true.

xattr /Users/rodion/Downloads/Clover.app 
com.apple.quarantine
xattr -rc  /Users/rodion/Downloads/Clover.app 
App is OK, i need xattr to start it too, macOS Sierra
one more way is Ctrl+Open

vectorsigma72, 2020-02-18:
> > You're right, and if fact I wanted to ensure the problem is not a damage archive for true.
> 
> xattr /Users/rodion/Downloads/Clover.app
> com.apple.quarantine
> xattr -rc /Users/rodion/Downloads/Clover.app
> App is OK, i need xattr to start it too, macOS Sierra
> one more way to run is "Ctrl+Open"

Looks like this is set after the download with Safari. An old app invoke the default broswer to download an update, so I can fix it while downloading within the app itself (writing a few hundred lines of code for this) or with an additional pkg with postinstall script. I hate peoples think this is a broken program.

roddy20, 2020-02-18:
curl -LJO https://github.com/CloverHackyColor/CloverBootloader/releases/download/5104/Clover.app-v1.15-5104.zip
unzip Clover.app-v1.15-5104
it is not broken )

SergeySlice, 2020-02-19:
I replaced Clover.app-v1.15-5104 by Clover.app.zip in the Release 5104.
Hope it help someone.

closed 2020-02-19 by SergeySlice