CloverHackyColor/CloverBootloader #83: 5104 scan entries hang

axet, 2020-02-18:
Hello!

I'm using Asus ZenBook UM431DA-AM003 (Ryzen5+Vega8). Having boot hang at "scan entries". Here is debug.log. Seems like stuck at AppleSCM.

axet, 2020-02-18:
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4218046/debug.log)

axet, 2020-02-18:
Removing SMCHelper.efi result in hang with following:

```
1:802  0:003  === [ Dump SMC keys from NVRAM ] ==========================
1:893  0:091  no AppleSMC protocol

```

SergeySlice, 2020-02-18:
Why you have no HFS+ or APFS driver?
Is there NVRAM supported on your platform?

axet, 2020-02-18:
New system. I'm trying to install osx. First to run clover efi loader, then start usb osx installer. Right?

root@axet-laptop:/home/axet# hexdump  /dev/nvram 
0000000 0000 0000 0000 7b00 ff02 00ff 0000 0000
0000010 0000 0000 0000 0000 0000 0000 0000 0000
0000020 7b02 ffff 0020 9d00 0000 0000 0000 0000
0000030 c200 0000 0201 4874 0000 0000 0000 0000
0000040 0000 0000 0000 0000 0000 0000 0000 0000
*
0000072

SergeySlice, 2020-02-19:
Should be as follow
~~~
8:413  0:002  === [ Dump SMC keys from NVRAM ] ==========================
8:466  0:052  found AppleSMC protocol
8:483  0:017  Registered 25 SMC keys
8:486  0:002  === [ ScanLoader ] ========================================

~~~

axet, 2020-02-20:
Since I see "scan entries" on a screen I assume "ScanLoader" hangs and never logs its entry. So, It is a scanloader unable to handle this hardware?

axet, 2020-02-20:
When I'm using Clover_v2.5k_r5104.pkg over Mojave prepared usb stick I see only blackscreen.

axet, 2020-02-26:
This hardware using nvme (NVM Express (NVMe) or Non-Volatile Memory Host Controller Interface Specification) for harddrive. Under linux is appears as /dev/nvme0 instead of /dev/sda

```
[    1.056885] nvme nvme0: pci function 0000:02:00.0
[    1.275278] nvme nvme0: 16/0/0 default/read/poll queues
[    1.277844]  nvme0n1: p1 p2 p3 p4
```

SergeySlice, 2020-03-18:
> Since I see "scan entries" on a screen I assume "ScanLoader" hangs and never logs its entry. So, It is a scanloader unable to handle this hardware?

Scanloader is able.

SergeySlice, 2020-03-18:
Why you have no HFS+ or APFS driver?

axet, 2020-03-18:
I drop all drivers to see working screen. No luck. I can add it back, it not going to help, right?

SergeySlice, 2020-03-20:
Test revision 5106 and upload debug.log.

axet, 2020-03-20:
Seems no changes

[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4358270/debug.log)

SergeySlice, 2020-03-20:
Add EmuVariableUEFI.efi into drivers/UEFI

axet, 2020-03-20:
Mojave usb drive failed to boot installer. Instead it shows CLOVER screen with no Mac Installer boot option (only windows and linux). I created Mojave drive with:

    sudo /Applications/Install\ macOS\ Mojave.app/Contents/Resources/createinstallmedia --volume /Volumes/MyVolume

Then run Clover_r5106.pkg on it with most drivers required (fat, EmuVariableUEFI, filevault).

axet, 2020-03-20:
```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *7.8 GB     disk2
   1:                        EFI EFI                     209.7 MB   disk2s1
   2:                  Apple_HFS Install macOS Mojave    7.5 GB     disk2s2
```

vectorsigma72, 2020-03-20:
I think is fixed with 5106

closed 2020-04-13 by SergeySlice