CloverHackyColor/CloverBootloader #84: macOS Install Prebooter not showing up

mcproj, 2020-02-23:
macOS 10.15.4 beta 1, Clover version 5104
The entry "macOS Install Prebooter" for updating does not show up after reboot for 10.15.4b2 update
It showed up fine on previous macOS versions

headkaze, 2020-02-28:
I managed to install macOS 10.15.4 Beta 1 but I can't install Beta 2 or 3. When performing an update and rebooting the Clover log shows the 'macOS - Data' partition listed but it doesn't show up in Clover's UI. The only option is selecting macOS which just boots to the previous version.

I also tried rolling back to Clover r5100 but it's still not showing up.

Attached is my Clover log after an install reboot.
[CLOVER.log](https://github.com/CloverHackyColor/CloverBootloader/files/4265744/CLOVER.log)

headkaze, 2020-03-11:
Same issue with macOS 10.15.4 Beta 4 and 5 and Clover 5105. Can't perform an update anymore.

norton287, 2020-03-11:
Workaround for me was to reinstall macOS 10.15.3 and then do the beta update.  I had to do this for the last two beta updates.

headkaze, 2020-03-12:
I'm wondering if we can force Clover to show the 'macOS Install' boot entry using the GUI/Custom/Entries option? The weird thing is the Clover log doesn't show it listed at all. I've tried adding the 'macOS - Data' Volume (including by GUID).

Maybe I need to add a 'Path' entry (Location to scan for the efi loader)? Does the macOS installer use a special efi loader?

headkaze, 2020-03-18:
@SergeySlice do you have any suggestions other than reinstalling macOS? Do you have any ideas why the 'macOS Install' mount is not showing when performing an update and rebooting to Clover? Is there a way we can force it to show?

norton287, 2020-03-19:
Anyone had any luck installing Beta 6?  I get error messages before the boot process starts.  Here is a pic of the last lines of the error messages
[image]

headkaze, 2020-03-19:
I ran:
``sudo softwareupdate -i -a
cd "/Volumes/macOS Base System/Install macOS Catalina Beta.app/Contents/MacOS"
sudo ./InstallAssistant
``
But it rolled me back to macOS 10.15.1 Beta 2 (19B77a)!

norton287, 2020-03-19:
@headkaze lol but good try though

headkaze, 2020-03-19:
So I just did an update from the App Store to macOS 10.15.3 (19D76) and will leave it there until there's a solution.

norton287, 2020-03-24:
This is fixed in latest clover release

headkaze, 2020-03-25:
Using Clover r5107 and updating to 10.15.4 and it does show the "Boot macOS Install from macOS - Data" message but on the second reboot it shows "macOS could not be installed on your computer. The installer resources were not found. Quit the installer to restart your computer and try again."

norton287, 2020-03-25:
Strange had no issues on my two hacks seemed like it went pretty smooth.  Maybe still something things in clover to look into.

daemeh, 2020-03-25:
Also had the issue with Clover r5106 - after starting an update, the macOS Install - Data option wouldn't appear anymore. I followed headkaze's example and downloaded the 10.5.4 installer, and the "Boot macOS Install from macOS - Data" option appeared in the Clover boot menu, it worked fine both reboots. Only issue was at the end of the install, it said something about "bless" being unable to set the boot disk. After the reboot, the "Boot macOS Install from macOS - Data" option was still there but I picked the normal macOS option and it booted into the updated macOS(19E266). Not sure if the other option will still be there, but I don't mind, I'd rather always see it than not see it at all :)

headkaze, 2020-03-25:
@daemeh the "bless" error was what I got on my second Hackintosh. The only problem seems to be that the "Boot macOS Install from macOS - Data" always shows now but as you say I'd rather that than it not show at all.

mcproj, 2020-03-26:
macOS Preboot option still doesn't show up with the latest version of Clover(Clover version 5107 + macOS 10.15.4 beta 5). My system partitions are FileVault enabled. 
I found this related line of code: https://github.com/CloverHackyColor/CloverBootloader/blob/84306ecbd89f1dd8c0dfe3d333394391c6a8020f/rEFIt_UEFI/entry_scan/loader.cpp#L1217
I've successfully updated from beta 1 to 5 by using the method provided by norton287. I mounted the Preboot partition and saw that there was a folder for entry "macOS Install from Preboot" inside, along with the one for "FileVault from Preboot" entry. I backed it up before I clicked the reboot button on 10.15.3 to update to beta 5. When it came to update to beta 6 I mounted that Preboot partition again and only found "FileVault" entry folder in it. I restored that "macOS Install" entry folder to its original place and then the "macOS Install from Preboot" entry showed up but when I tried to boot from that entry it said that it couldn't find "prelinkedkernel" file and failed to boot. Seems like the update files are not placed correctly by Apple's update process. I have no idea whether I should disable FileVault to make the update to work.

mcproj, 2020-03-26:
Seems like downloading macOS install.app from mac App Store makes me able to update to 10.15.4 GM version. So this issue is just affecting updates directly from Preferences.app(like those beta updates)

norton287, 2020-03-31:
I had to download the installer for my 10.15.4 beta 6 third box.  It updated successfully then.  

Then I tried to update my second box to 10.15.5 beta 1 and its the same thing no PreBooter volume or option to install it after rebooting.

norton287, 2020-04-01:
Read a post on insanelymac that gave me what I needed.  First boot to clover and press F11, then boot to macOS.  Then download and run the installer and reboot and the install option will be there.  I just did that on two boxes and it worked fine.

SergeySlice, 2020-05-18:
Is the issue true for Clover 5117?

mcproj, 2020-05-18:
> Read a post on insanelymac that gave me what I needed. First boot to clover and press F11, then boot to macOS. Then download and run the installer and reboot and the install option will be there. I just did that on two boxes and it worked fine.

Thanks for your method. This works for me. 

> Is the issue true for Clover 5117?

See method provided by norton287. Press F11 on Clover boot screen to clear your nvram, and then try update. I've yet to update to 5117, but it worked for me on 5109.

closed 2020-05-18 by mcproj

daemeh, 2020-05-28:
Just a tip for those trying to download the installer - if you can't find the macOS version you want in the AppStore, you can download it manually using:
`softwareupdate --fetch-full-installer --full-installer-version 10.15`
More about that here https://derflounder.wordpress.com/2019/10/28/downloading-macos-installers-with-updated-signing-certificates-on-macos-catalina/