CloverHackyColor/CloverBootloader #85: Add Better README.md

circutrider21, 2020-03-04:
## Explanation 
This Is a situation because I am a OS Developer for Peeled Fruit Studios. When I first came to this repository, I had no idea what this repository did a all. Neither did i know how to use this Bootloader. Most importantly, I won't understand until you write a README that explains how to use this Bootloader. I am currently working on Storm-OS which is our latest project, and I am looking for an alternative to GNU GRUB. I would be glad to use your Bootloader, but i don't know how to use it.

CMMChris, 2020-03-05:
Clover is for running macOS on PC hardware. Why would you use such a bloated piece of software to not boot macOS but other OSes instead? Doesn't make any sense.

circutrider21, 2020-03-09:
Now I understand **but other people won't unless you add a README**

xstable, 2020-04-03:
I was piped to this repo by https://sourceforge.net/projects/cloverefiboot/files/
Can you tell me where to get the cloverefiboot here and how to use it?

A README with some basic-informations would definitely be a good Idea!

circutrider21, 2020-04-13:
Exactly, that is how i feel

closed 2020-04-29 by asava

RandomDSdevel, 2020-05-01:
@asava:  What resolved this issue, allowing you to close it?

asava, 2020-05-01:
We prefer not to have documentation as part of open issues. It was tagged as documentation, and will be taken care of, when we get it to it. Thank you for your concern.