CloverHackyColor/CloverBootloader #86: 5104: many times on startup clover is not detected by the uefi bios

mippol, 2020-03-08:
After upgrading to version 5104 many times clover is not detected and I have to restart,
If I replace the AMD RX480 graphics card with an NVIDIA GTX780 there is no problem.
Thank You!

SergeySlice, 2020-03-08:
Please provide log and other information to prove the issue.

mippol, 2020-03-08:
0:100  0:100  MemLog inited, TSC freq: 3696527640
0:100  0:000  CPU was calibrated with RTC
0:100  0:000  
0:100  0:000  Now is 08.03.2020,  09:35:30 (GMT)
0:100  0:000  Starting Clover revision: 5104 (master, commit 60b6eb5c3) on American Megatrends EFI
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED --conf=/Users/sergey/src/CloverHackyColor/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SelfDevicePath=PciRoot(0x0)\Pci(0x1D,0x0)\Pci(0x0,0x0)\NVMe(0x1,9B-10-B4-91-56-38-25-00)\HD(1,GPT,DB5FEAFF-E92B-496B-A05E-323200F7841A,0x28,0x64000) @8A533518
0:100  0:000  SelfDirPath = \EFI\BOOT
0:100  0:000  Clover : Image base = 0x7DC07000
0:100  0:000  g_str = g_str:foobar
0:100  0:000  g_str2 = g_str:foobar2
0:100  0:000  global_str1 = global_str1
0:100  0:000  global_str2 = global_str2
0:100  0:000  SimpleTextEx Status=Success
0:100  0:000  === [ Get Smbios ] ========================================
0:100  0:000  Type 16 Index = 0
0:100  0:000  Total Memory Slots Count = 2
0:100  0:000  Type 17 Index = 0
0:100  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:100  0:000  SmbiosTable.Type17->Size = 16384MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 ChannelA-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = Corsair
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = XXXXXXXXXXXXXXXXXXXX 
0:100  0:000  Type 17 Index = 1
0:100  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:100  0:000  SmbiosTable.Type17->Size = 16384MB
0:100  0:000  SmbiosTable.Type17->Bank/Device = BANK 2 ChannelB-DIMM1
0:100  0:000  SmbiosTable.Type17->Vendor = Corsair
0:100  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:100  0:000  SmbiosTable.Type17->PartNumber = XXXXXXXXXXXXXXXXXXXXX  
0:100  0:000  Boot status=0
0:100  0:000  Running on: 'System Product Name' with board 'ROG STRIX H370-I GAMING'
0:100  0:000  === [ GetCPUProperties ] ==================================
0:100  0:000  CPU Vendor = 756E6547 Model=906EA
0:100  0:000   The CPU supported SSE4.1
0:100  0:000  BrandString = Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz
0:100  0:000   The CPU supported turbo
0:100  0:000  MSR 0x35               6000C
0:100  0:000   TSC/CCC Information Leaf:
0:100  0:000    numerator     : 308
0:100  0:000    denominator   : 2
0:100  0:000   Calibrated ARTFrequency: 24003426
0:100  0:000   Rounded ARTFrequency: 24000000
0:100  0:000  MSR 0xE2 before patch 1E000000
0:100  0:000  MSR 0xCE              08080838_F1012500
0:100  0:000  corrected FLEX_RATIO = E0000
0:100  0:000  FSBFrequency = 99 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:100  0:000  MaxDiv/MinDiv: 37.0/8
0:100  0:000  Turbo: 44/45/46/47
0:100  0:000  Features: 0xBFEBFBFF
0:100  0:000  Threads: 12
0:100  0:000  Cores: 6
0:100  0:000  FSB: 100 MHz
0:100  0:000  CPU: 3663 MHz
0:100  0:000  TSC: 3663 MHz
0:100  0:000  PIS: 100 MHz
0:100  0:000  ExternalClock: 25 MHz
0:100  0:000  === [ GetDevices ] ========================================
0:100  0:000  GOP found at: PciRoot(0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/AcpiAdr(0x80010300)
0:100  0:000  PCI (00|00:00.00) : 8086 3EC2 class=060000
0:100  0:000  PCI (00|00:01.00) : 8086 1901 class=060400
0:100  0:000  PCI (00|01:00.00) : 1002 67DF class=030000
0:100  0:000   - GOP: Provided by device
0:100  0:000   - GFX: Model=AMD Radeon RX 480 (ATI/AMD)
0:100  0:000  PCI (00|01:00.01) : 1002 AAF0 class=040300
0:100  0:000   - HDMI Audio: 
0:100  0:000  PCI (00|00:14.00) : 8086 A36D class=0C0330
0:100  0:000  PCI (00|00:14.02) : 8086 A36F class=050000
0:100  0:000  PCI (00|00:16.00) : 8086 A360 class=078000
[... 432 lines cut ...]
6:497  0:002  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Other
6:497  0:000  ->Extra kext: EFI\CLOVER\kexts\Other\BrcmPatchRAM3.kext (v.2.5.1)
6:565  0:067  ->Extra kext: EFI\CLOVER\kexts\Other\USBPorts.kext (v.1.0)
6:568  0:003  ->Extra kext: EFI\CLOVER\kexts\Other\BrcmBluetoothInjector.kext (v.2.5.1)
6:625  0:056  ->Extra kext: EFI\CLOVER\kexts\Other\VirtualSMC.kext (v.1.1.1)
6:635  0:010  ->Extra kext: EFI\CLOVER\kexts\Other\BrcmFirmwareData.kext (v.2.5.1)
6:642  0:007  ->Extra kext: EFI\CLOVER\kexts\Other\CPUFriend.kext (v.1.2.0)
6:650  0:007  ->Extra kext: EFI\CLOVER\kexts\Other\SMCProcessor.kext (v.1.1.1)
6:658  0:007  ->Extra kext: EFI\CLOVER\kexts\Other\RealtekRTL8111.kext (v.2.2.2)
6:664  0:006  ->Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.3.7)
6:673  0:009  ->Extra kext: EFI\CLOVER\kexts\Other\SMCSuperIO.kext (v.1.1.1)
6:680  0:007  ->Extra kext: EFI\CLOVER\kexts\Other\IntelMausi.kext (v.1.0.2)
6:687  0:006  ->Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.4.7)
7:231  0:544  ->Extra kext: EFI\CLOVER\kexts\Other\AirportBrcmFixup.kext (v.2.0.6)
7:249  0:017  ->Extra kext: EFI\CLOVER\kexts\Other\BT4LEContinuityFixup.kext (v.1.1.4)
7:258  0:009  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.4.2)
7:267  0:008  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Off
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10_normal
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15_normal
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.3
7:267  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.3_normal
7:269  0:001  SetScreenResolution: 3440x1440 - already set
7:269  0:000  Custom boot is disabled

mippol, 2020-03-08:
> Please provide log and other information to prove the issue.

[log.txt](https://github.com/CloverHackyColor/CloverBootloader/files/4303049/log.txt)

SergeySlice, 2020-03-10:
What is not detected?
`6:406  0:000  === [ ScanLoader ] ========================================
6:406  0:000  - [01]: 'EFI'
6:408  0:001          AddLoaderEntry for Volume Name=EFI
6:408  0:000          skipped because entry is hidden
6:408  0:000  - [03]: 'PreBoot'
6:408  0:000          AddLoaderEntry for Volume Name=PreBoot
6:410  0:001      Check if volume Is Hibernated:
6:410  0:000      Check sleep image 'by signature':
6:410  0:000      using default sleep image name = \private\var\vm\sleepimage
6:410  0:000      sleepimage not found -> Not Found
6:410  0:000      hibernated: no - sign
6:413  0:002  - [04]: 'Macintosh NVMe - Data'
6:414  0:001  - [05]: 'Recovery'
6:414  0:000          AddLoaderEntry for Volume Name=Recovery
6:416  0:002  - [06]: 'VM'
6:416  0:000  - [07]: 'Macintosh NVMe'
6:420  0:003          AddLoaderEntry for Volume Name=Macintosh NVMe
6:421  0:001      Check if volume Is Hibernated:
6:421  0:000      Check sleep image 'by signature':
6:422  0:000      using default sleep image name = \private\var\vm\sleepimage
6:422  0:000      sleepimage not found -> Not Found
6:422  0:000      hibernated: no - sign
`

Jacksonzwang, 2020-03-27:
Did you mean that there is no display if you only insert with your AMD graphics card? If it is, try to disable CSM in UEFI and delete CsmVideoDxe.efi file. I just guess the conclusion from the following code in your log file:
```
0:112  0:000  Loading CsmVideoDxe.efi  status=Success
0:112  0:000   - driver needs connecting
```

mippol, 2020-03-27:
Hi !
'M sorry for delay...but..after some experiment, solved with motherboard reset: completely OnBoard/External Graphic issue discovered!
Thank for the help and 'm sorry for the mistake!!!

closed 2020-03-27 by mippol