CloverHackyColor/CloverBootloader #89: The icons becomes strange.

mifjpn, 2020-03-12:
Thanks for this wonderful app.
I download clover r5105.pkg and install it.
Icons of clover theme become strange.
I use Minimalism theme.
So I change another theme, but the Icons become strange.
Is there any way to cure it?

pit512, 2020-03-12:
Same here.
I will try to take a screenshot...

pit512, 2020-03-12:
Look at those Screenshots (Before and After)...

[image]
[image]

vectorsigma72, 2020-03-12:
Please use Clover.app to optimize the theme:
[image]
some theme needs that because the author forgot to do some step..

pit512, 2020-03-12:
OK, will try...
Thank you.

LeeBinder, 2020-03-12:
hey @vectorsigma72 , thank you for the optimize hint. I am experiencing another icon related bug in the Clover start-up GUI ever since build r5105 which can't be solved with the theme optimization. Rather than opening a new topic I opt to file it here.

The new bug in r5105 is that it disregards a volume's .VolumeIcon.icns file at the root of the volume, even if 'Custom Icons' is enabled in the GUI section. That even applies to the embedded theme.

Custom 128x128 .png icons set via GUI/ Custom Entries/ Title are still displayed, so I thought, let's try something: in the entries where the 'Title' fields were left empty so far because Clover GUI would get these entries' icons automatically from the respective volume's .VolumeIcon, I pasted .VolumeIcon.icns into the custom entries title fields, saved, but on reboot the custom volume icons were still not back (because .VolumeIcon files are not PNG but a proprietary format).

It would be really neat if you or someone from the Clover dev team could look into the new C++ code and find the culprit spot in the code which suddenly leads to disregarding .VolumeIcon.icns files.

Thanks to you and the team for Clover !!!

pit512, 2020-03-13:
> Please use Clover.app to optimize the theme:
> some theme needs that because the author forgot to do some step..

Here is the result after optimising Minimalism theme...  

[image]

I believe new Clover (5105) can't read “real” ICNS files anymore, it can only read ICNS files which are in fact PNG files with an .icns extension...

vectorsigma72, 2020-03-13:
> hey @vectorsigma72 , thank you for the optimize hint. I am experiencing another icon related bug in the Clover start-up GUI ever since build r5105 which can't be solved with the theme optimization. Rather than opening a new topic I opt to file it here.
> 
> The new bug in r5105 is that it disregards a volume's .VolumeIcon.icns file at the root of the volume, even if 'Custom Icons' is enabled in the GUI section. That even applies to the embedded theme.
> 
> Custom 128x128 .png icons set via GUI/ Custom Entries/ Title are still displayed, so I thought, let's try something: in the entries where the 'Title' fields were left empty so far because Clover GUI would get these entries' icons automatically from the respective volume's .VolumeIcon, I pasted .VolumeIcon.icns into the custom entries title fields, saved, but on reboot the custom volume icons were still not back (because .VolumeIcon files are not PNG but a proprietary format, but eh, worth a try).
> 
> It would be really neat if you or someone from the Clover dev team could look into the new C++ code and find the culprit spot in the code which suddenly leads to disregarding .VolumeIcon.icns files.
> 
> Thanks to you and the team for Clover !!!

Hi this is indeed another problem. I'll tell that to Slice

SergeySlice, 2020-03-13:
Yes, we should keep icns support for VolumeIcon.

vectorsigma72, 2020-03-13:
> > Please use Clover.app to optimize the theme:
> > some theme needs that because the author forgot to do some step..
> 
> @vectorsigma72, here is the result after optimising the Minimalism theme using Clover.app...
> 
> [image]
> 
> About the bootloader itself, I believe release 5105 can't read “real” ICNS files anymore, it can only read ICNS files which are in fact PNG files with an .icns extension...

I'll do a better algorithm as it actually optimize all the images inside the theme directory:

now:
1. scan for all non png images and convert them if a need be.
2. convert all images to png-8

then:
1. scan for non png images and convert them if a need be.
2. convert all icns inside the icon sub directory to png-8

in fact the icon issue is solved, but the backgroung doesn't need the optimization.

vectorsigma72, 2020-03-13:
> @vectorsigma72, here is the result after optimising the Minimalism theme using Clover.app...
>
> [image]
>
> About the bootloader itself, I believe release 5105 can't read “real” ICNS files anymore, it [...]

Can I have your original theme?

LeeBinder, 2020-03-13:
> Yes, we should keep icns support for VolumeIcon.

great, thanks Slice. I'm confident the code to be ported to C++ from r5104 isn't monumental. If there is a need for beta testing, I'll volunteer :)

pit512, 2020-03-14:
> > > Please use Clover.app to optimize the theme:
> > > some theme needs that because the author forgot to do some step..
> > 
> > 
> > @vectorsigma72, here is the result after optimising the Minimalism theme using Clover.app...
> > ![screenshot2]
> > About the bootloader itself, I believe release 5105 can't read “real” ICNS files anymore, it can only read ICNS files which are in fact PNG files with an .icns extension...
> 
> Can I have your original theme?

Here, the original and the one modified by myself...

[Minimalism_Original.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4332032/Minimalism_Original.zip)
[Minimalism.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4332033/Minimalism.zip)

mifjpn, 2020-03-14:
Dear pit512
Thanks for the reports.
Me too.
I hope that the problem is solved.

mifjpn, 2020-03-15:
In minimalism, the size of the icons seems to be bad.
I tried to create alternative themes by referring to some themes.
If you like, try using that theme.
[https://mifmif.ddo.jp/pcpc/public/minimalism2.zip](https://mifmif.ddo.jp/pcpc/public/minimalism2.zip)
I was helped by Optimize of Colver.app in making it. Thank you.

pit512, 2020-03-15:
OK, thank you mifjpn

LeeBinder, 2020-03-22:
thank you @SergeySlice or @vectorsigma72 or whoever from the Clover dev team fixed the .VolumeIcon.icns regression in the latest r5106 !!! 👍 🥇 💯

Anard, 2020-03-26:
Hi all.
I've exactly the same problems with Clover v.5106
Here are my theme's base and optimized versions... : https://github.com/Anard/Sketched_CloverTheme

pit512, 2020-03-26:
Hi Anard,
Try to convert your ICNS files to PNG. That should fix it.

Anard, 2020-03-26:
Thx for your answer Pit,
But I tried to convert my os_xxx.icns to png in 1024x1024x32 or 128x128x32, still with the "unknown icon" symbol in Clover.
After using the optimize tool from Clover.app, icons were found but like [vectorsigma72](https://github.com/CloverHackyColor/CloverBootloader/issues/89#issuecomment-598902717), they were very large and theme is still unusable...

pit512, 2020-03-26:
You can try to edit them to make them smaller (128x128).
Also, try to temporarily replace one of your ICNS file with this one, just to test and see...
[os_mac.icns.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4388376/os_mac.icns.zip)

pit512, 2020-03-26:
just a thought: try to delete all ICNS files in your icons folder and see what happens...

Anard, 2020-03-27:
Hi back. No improvement with only png 128x128 files. But I tried to delete all os_(mac|moj|yos|cap|snow|leo|lion).png and replace them by your os_mac.icns and it works.
How have you build this image ?

Anard, 2020-03-27:
OK. II saved my icons in PNG format, 128x128px but renamed it with .icns extension and it's now OK.
For cursor, it's a std PNG, 32x32px with ..png extension.
Anyway, it's a little strange...

pit512, 2020-03-28:
> Anyway, it's a little strange...

Absolutely, as pointed out in post 7, I believe Clover (starting from release 5105) can no longer read real ICNS files, it can only read ICNS files which are in fact PNG files with an .icns extension...

pit512, 2020-04-06:
New GUI issue with 5109...
This theme is supposed to be white on black.
[image]

mifjpn, 2020-04-12:
Should be raised as an issue for the next version.

closed 2020-04-12 by mifjpn