CloverHackyColor/CloverBootloader #90: Clover hangs on Qemu

tensor5, 2020-03-12:
I'm new to Clover, so my way of running Clover might be incorrect.
I extract the `EFI` folder from `Clover-5105-X64.iso` into a GPT partitioned image disk named `clover.img`. The I run:

```sh
qemu-system-x86_64 -enable-kvm -m 2G -machine q35 -cpu host -drive if=pflash,format=raw,readonly,file=/usr/share/ovmf/x64/OVMF_CODE.fd -drive if=pflash,format=raw,file=OVMF_VARS.fd -drive if=virtio,file=clover.img,format=raw
```

From the EFI shell I can see the drive content, but when I type `fs0:\EFI\BOOT\BOOTX64.EFI` and it hangs. The OVMF files are provided by an Arch Linux package. Am I missing something?

SergeySlice, 2020-03-13:
Never hands up! There are thousands nuance how to start QEMU.
One of possible way is to follow our way in the folder "qemu" in the Clover sources.

tensor5, 2020-03-16:
It works if I set `-cpu Penryn`, but doesn't with `-cpu SandyBridge` or `-cpu host` (in my case host is an IvyBrigde). Any idea why?

SergeySlice, 2020-03-18:
Because QEMU is not perfect.

VictorGamer072YT, 2020-03-23:
> 
> 
> It works if I set `-cpu Penryn`, but doesn't with `-cpu SandyBridge` or `-cpu host` (in my case host is an IvyBrigde). Any idea why?

You probably want to use OpenCore to use the IvyBridge as host.

tensor5, 2020-03-23:
@VictorGamer072YT I tried OpenCore but it has the same problem.

SergeySlice, 2020-04-19:
I see no Clover issue. It is user issue how to use QEMU.

closed 2020-04-19 by SergeySlice