CloverHackyColor/CloverBootloader #91: Clover Updater goes to the wrong web page to download r5105

najeebanwer, 2020-03-12:
When downloading the r5105 version of Clover, the updater goes to this URL:

https://github.com/CloverHackyColor/CloverBootloader/releases/download/5105/Clover_v2.5k_r5105.pkg

the download is actually located at:
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5105/Clover_r5105.pkg

SergeySlice, 2020-03-13:
You have to update the Updater.

najeebanwer, 2020-03-14:
If you mean the Clover updater that is part of the Clover Preference Pane, I always upgrade the PrefPane when installing a new version of Clover.
[image]
If there is something else I need to do to update the Updater, please let me know.

vectorsigma72, 2020-03-18:
> If you mean the Clover updater that is part of the Clover Preference Pane, I always upgrade the PrefPane when installing a new version of Clover.
> [image]
> If there is something else I need to do to update the Updater, please let me know.

Bro, if you already Installed the last revision .... how can find a newer one if doesn't exist yet?

najeebanwer, 2020-03-18:
@vectorsigma72 
Bro, the intent of this ticket is to report that there was a problem with updating from r5104 to r5105 because the Clover updater was attempting to download from the wrong URL. I was able to update to r5105 by manually downloading from the correct URL. 
The advice I was given is that "You have to update the Updater". I don't think that it is unreasonable to ask if just upgrading the Clover PrefPane is all that is required to "update the updater" so that this issue does not happen during the next Clover update.

vectorsigma72, 2020-03-18:
> @vectorsigma72
> Bro, the intent of this ticket is to report that there was a problem with updating from r5104 to r5105 because the Clover updater was attempting to download from the wrong URL. I was [...]

We discussed about this already [here](https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2710771) 15 days ago, so nothing to fix as was taken into account for the switch. In future will be no more this problem at every vx.xk.

najeebanwer, 2020-03-18:
OK. Ticket Closed.

closed 2020-03-18 by najeebanwer