CloverHackyColor/CloverBootloader #92: Add ACPI Patch to support QEMU Virtio-net

a17uk, 2020-03-13:
Apple added VirtIO support since Catalina. It is an excellent feature for KVM based macOS. I tried VirtIO block device and it works well for me. But the Virtio-net device, even macOS can identify and drive it, but it seems to change its mac address every time booting from EFI. Thus lose the connection. As mentioned by Leoyzen in this topic https://www.insanelymac.com/forum/topic/340797-virtio-net-support-added-in-catalina/

The only difference s/he has found in ACPI is "IOFeature" variable, it works when set to 0x38, and random mac address with 0x08.

Is that possible to make a patch and add it to Clover ACPI patch options? Because of the insane performance of the VirtIO-Net, it would be great for KVM-macOS users. I tried but my KVM-macOS build does not have an ACPI folder in the EFI partition, it seems the only way for me to edit the DSDT options is by using the Clover configurator.

SergeySlice, 2020-03-13:
I am sorry but I understand nothing.
Why do you have no an ACPI folder somewhere? mkdir and be happy!
What DSDT option you are going to edit with CC?
What is IOFeature variable? Where does she live?
What is mac address in your world? Usually it is 6 bytes sequence? Not just 0x8 as you said.

a17uk, 2020-03-14:
The 0x8 is the IOFeature value of DSDT, not the mac address. It seems my the ACPI of my build is entirely managed by CC. No ACPI folder no .aml files, so I have no way to manually modify them.

This image is shown in that topic I mentioned. Just change the IOFeature Value from 0x8 to 0x38.
[image]

SergeySlice, 2020-03-15:
The value IOFeatures produces by the driver not by DSDT.

a17uk, 2020-03-16:
Which driver? I tried to find, it seems /System/Library/Extensions/IONetworkingFamily.kext/Contents/MacOS/IONetworkingFamily is related to the "IOFeatures" property.

asava, 2020-05-01:
This is not directly related to Clover. You probably need to find which binary patch you need for your problem and update it in your config.plist

closed 2020-05-01 by asava