CloverHackyColor/CloverBootloader #93: R5105 Dose not boot automatically

Razor004, 2020-03-14:
R5105 Dose not boot automatically the default boot volume that i specify , had to downgrade to r5104 to fix the problem .

SergeySlice, 2020-03-15:
No, they are the same.
Go to macOS, choose System Preferences -> StartupDisk and make restart from here.
Next restart will be automatic by timeout.

Razor004, 2020-03-16:
I did that and the first restart works great then the second one doesn't boot by timeout .

Razor004, 2020-03-17:
I have a update about this issue , on my pc usually dual boot with windows 10 so I have multiple drivers ( 2 ssds) on 1 ssd I have Mac OS 10.15.3 and on the other ssd I have windows 10 , if I unplug the ssd with windows, clover boots automatic by timeout , and if I plug it in clover sees the windows partitions and doesn't boot by timeout , I don't have this issue with 5104 , sorry for my bad English .

SergeySlice, 2020-03-18:
5104 and 5105 are the same with this function.

SergeySlice, 2020-03-26:
User's mistake

closed 2020-03-26 by SergeySlice

brunoaduarte, 2020-08-31:
Not user's mistake. Problem still exists if user has another SSD with Windows 10 installed.