CloverHackyColor/CloverBootloader #96: R5105 Unable to boot linux (kubuntu)

pagimamaun, 2020-03-17:
Hello
I use clover for multi-boot: Catalina, Windows 10, kubuntu 19,10
After this update, i was unable to start kubuntu with clover
Grub starting fine but after starting kubuntu i'm stuck on black screen
With verbose loading is stuck on "loading ramdisk"
Thanks for your help

SergeySlice, 2020-03-18:
Any information? How to reproduce?

asava, 2020-03-26:
I can confirm how this issue can be produced, but it is not caused by Clover.
You are using OcQuirks, right?
This is caused by OpenRuntime.efi (or FwRuntimeServices.efi in older versions).

Steps to reproduce:
1. boot to shell (without Clover at all)
2. load OpenRuntime.efi (or FwRuntimeServices.efi in older versions).
3. start grubx64.efi
The same behavior you are describing will occur.

If this is the case, please report this to the OpenCore team.

pagimamaun, 2020-04-06:
Hello
Thank you for your response
The problem was reported to open core team
A new version of ocquirks was done and now all is ok with kubuntu boot
Thanks for your help

closed 2020-04-06 by pagimamaun