CloverHackyColor/CloverBootloader #99: Download URL on upgrade is wrong - 404

Macrisu, 2020-03-20:
On my Hackintosh, I get warning for the upgrade, but the link is wrong

ex:
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5106/Clover_r5106.pkg

The correct link is:
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5106/Clover.app_v1.16_r5106.pkg.zip

The problem started a few version ago.

pit512, 2020-03-20:
You're almost right, lol, the correct link for Clover Bootloader should be :
https://github.com/CloverHackyColor/CloverBootloader/releases/download/5106/Clover_r5106.zip

vectorsigma72, 2020-03-20:
> You're almost right, lol, the correct link for Clover Bootloader should be :
> https://github.com/CloverHackyColor/CloverBootloader/releases/download/5106/Clover_r5106.zip

Sorry no, the upload was wrong: https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2712993

closed 2020-03-22 by SergeySlice