CloverHackyColor/CloverBootloader #101: r5107 boots to black screen

sprtm, 2020-03-25:
I updated to r5107 and now I get nothing but a black screen when Clover loads. I also tried to downgrade to r5102, r5103 and r5106, but same issue.

One time the GUI loaded, but stuck after I chose the startup disk. At some boots the theme startup chime plays, but still a black screen.

I have now downgraded to Dids' r5103 Clover build. For some reason Dids' automated Clover builds works without issues on my machine (but r5107 isn't available there).

[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/4381091/debug.log)

gianfigueroa, 2020-03-25:
I have the same error, I need any fast solutions, can I downgrade clover??

vectorsigma72, 2020-03-25:
> I updated to r5107 and now I get nothing but a black screen when Clover loads. I also tried to downgrade to r5102, r5103 and r5106, but same issue.
>
> One time the GUI loaded, but stuck after I chose [...]

66 Themes??

sprtm, 2020-03-25:
Dunno. Guess from CloverThemeManager. Will it affect performance?

gianfigueroa, 2020-03-25:
How I can downgrade clover ??, please

sprtm, 2020-03-25:
So, I trashed all themes except the one I use, and now r5107 boots!

gianfigueroa, 2020-03-25:
How you trashed??

sprtm, 2020-03-25:
Just delete them from /Volumes/EFI/EFI/CLOVER/themes.

SergeySlice, 2020-03-25:
Sorry, but official Clover supports 50 themes. If more then crash.

closed 2020-03-25 by SergeySlice

gianfigueroa, 2020-03-25:
> Just delete them from /Volumes/EFI/EFI/CLOVER/themes.

For where??

Shell efi??

vectorsigma72, 2020-03-25:
> > Just delete them from /Volumes/EFI/EFI/CLOVER/themes.
> 
> For where??
> 
> Shell efi??

Use Clover.app (look at releases) and open the ESP (EFI System Partition) and delete the themes manually (i.e. with the Finder).

gianfigueroa, 2020-03-25:
> > > Just delete them from /Volumes/EFI/EFI/CLOVER/themes.
> > 
> > 
> > For where??
> > Shell efi??
> 
> Use Clover.app (look at releases) and open the ESP (EFI System Partition) and delete the themes manually (i.e. with the Finder).

My system is not running always screen black, I can't access to finder

gianfigueroa, 2020-03-25:
or How I can se the log for detect anthoer error.I Never have been installed themes