CloverHackyColor/CloverBootloader #102: About apple warning about furute kext incompatibility

DMNerd, 2020-03-25:
Hello, I was wandering if you have noticed the situation with the newest macos update and if you are planning to do anything about it
[image]
Format: [image]

Jacksonzwang, 2020-03-27:
I guess it might be refer to the extension which is provided by applications such as "Intel Gadget" and so on. Rather than system-level extensions.

DMNerd, 2020-03-27:
Hopefully, tohugh there has been a debate about it. The question is still valid though. Is there any plan B for clover, just so users don`t have to switch to OC in the future

roddy20, 2020-03-27:
> Hopefully, tohugh there has been a debate about it. The question is still valid though. Is there any plan B for clover, just so users don`t have to switch to OC in the future

How is it related to Clover and OC ?

DMNerd, 2020-03-27:
> > Hopefully, tohugh there has been a debate about it. The question is still valid though. Is there any plan B for clover, just so users don`t have to switch to OC in the future
> 
> How is incompatible KEXT related to Clover and/or OC ?

I the way the kext injection works. Both clover and OC use fundamentaly different techniques to inject kexts on boot. All I wanted with this issue to start conversation about future of clover if apple starts blocking external kexts. That`s all

SergeySlice, 2020-04-19:
Apple will block old kexts created by old technology.
No problem nowadays so no issue.

closed 2020-04-19 by SergeySlice